nordicsemi/IOS-DFU-Library #73: Crash in LegacyDFUService

anlaital, 2017-01-11:
We just got a crash in our internal release from `LegacyDFUService`:

```
    func sendReset(onError report: @escaping ErrorCallback) {
        dfuControlPointCharacteristic!.send(Request.reset, onSuccess: nil, onError: report)
    }
```

It seems that `dfuControlPointCharacteristic` is nil when force-unwrapping under some circumstance.

http://crashes.to/s/36f2e68be4a

philips77, 2017-01-11:
I'll fix it right away.

philips77, 2017-01-11:
Which lib version are you using? Looks like I have already fixed this bug in 3.0.1 here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/commit/58bea539fea2474c20867ad0a4382a8a7cc90fcc
I don't see how could it crash there with this fix. From your stack trace it looks like it failed when Control Point characteristic hasn't been found, the defaultErrorCallback was called which eventually called LegacyDFUPeripheral.resetDevice() and this shouldn't get into sendReset unless this characteristic exists.

anlaital, 2017-01-11:
Thanks for the response! This has indeed been fixed in `3.0.1` and we are using `3.0.0`. I will try to uncover the reason blocking our update of the SDK that's caused by issue #69.

closed 2017-01-11 by anlaital