nordicsemi/IOS-DFU-Library #74: Can not update single hex file

tcao, 2017-01-13:
Tried to update the single .hex application, failed with error 103 (The init packet is required by the target device).
Tried the same .hex applications with Master Control Panel, all updated successfully. I guess MCP is using different DFU library.

Here's the Xcode console output:
V: Connecting to DfuTarg...
D: centralManager.connect(peripheral, options:nil)
Changed state to: Connecting
D: [Callback] Central Manager did connect peripheral
I: Connected to DfuTarg
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Legacy DFU...
I: Connected to DfuTarg
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A: Version number read: 0.8
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
Error 103: The init packet is required by the target device

philips77, 2017-01-13:
This is not a bug. Dfu bootloaders require init packet starting from SDK 7, as far as I remember. In the same SDK the DFU Version characteristic had been added.
As you see on the log above the DFU Version char was found and returned 0x0800 which means it's the newest legacy bootloader, SDK 8+, or something like that. At least that is for a bootloaders that weren't altered.
You may provide the init packet as dat file or use the Distribution Packet ZIP with all included.
It isn't also possible that it works on Android. The PC MCP app, depending on it's version, also supports ZIPs and sending a single hex file should be rejected by the bootloader in your case. 
Are you sure you have flashed the same bootloader for two trials?

tcao, 2017-01-13:
Yes, I'm using the same bootloader comes with nRF5 SDK 11, on nRF52 DK.
1) PC MCP version 3.10.0.14 can update single hex application (tried peripheral example blinky_s132, and ble_peripheral hrs s132_with_dfu) w/o problem.
2) IOSDFU Library's example iOS project failed with error 103 on both hex applications, but ZIP works fine.

philips77, 2017-01-13:
Perhaps the MCP is creating the init packet automatically in the background. You could check it by sniffing the transmission or logging it from the bootloader. I'll check if this is the case when I get to work.

tcao, 2017-01-13:
Thanks Philips. 
Is there any article/tutorial talking about logging, in particular boot loader logging?
I enabled RTT for the bootloader, just want to know where are the good code locations to generate the RTT outputs.

philips77, 2017-01-13:
For that you'd have to ask on DevZone on Nordic's website. I'm not an expert in firmware. I think you could print data received by Control Point char and look for 0x0200 and 0x0201 (begin and end of sending an init packet).

closed 2017-02-07 by mostafaberg