nordicsemi/IOS-DFU-Library #75: Can't find spec for 'Zip' module

kevinpk, 2017-01-17:
I'm trying to install the DFU pod in my project but get this error when running pod install:
`[!] Unable to find a specification for 'Zip (~> 0.6)' depended upon by 'iOSDFULibrary'`

Here is my Podfile:
`target 'DFU_test' do`
`  use_frameworks!`
`  pod 'iOSDFULibrary'`
`  # Pods for DFU_test`
`end`

mostafaberg, 2017-01-17:
You probably need to update your cocoapods repo, try this:
`pod repo update`
and then you may run
`pod install`

let me know how it goes

kevinpk, 2017-01-17:
Thanks for the suggestion @mostafaberg. I ended up having to delete, then recreate my specs repo:

`pod repo remove master`
`pod setup`

Then `pod install` finally worked!

closed 2017-01-17 by kevinpk