nordicsemi/IOS-DFU-Library #76: Problem including the DFU lib in another lib that is distributed over Carthage

AlexDM0, 2017-01-18:
Hi,

We're developing the Crownstone, which uses the NRF52. I'm developing the iOS lib and consumer app for the product. I'd like to include the Nordic DFU lib as part of my lib so I can wrap all Crownstone related operations in our native lib. This lib is distributed via Carthage and used by a number of different applications.

My lib's cartfile looks like this:

```
# Crypto related functions and helpers for Swift
github "krzyzanowskim/CryptoSwift" "0.6.1"

# We code in Asynchronous world. We promise. [http://promisekit.org/]
github "mxcl/PromiseKit"

# JSON Parser
github "SwiftyJSON/SwiftyJSON"

# DFU code from Nordic
github "NordicSemiconductor/IOS-Pods-DFU-Library"
```

I added a run script to my lib:

```
# required to build framework in xcode
rm -rf Carthage/Checkouts/IOS-Pods-DFU-Library/Example/Pods/
```

in order to avoid the issue posted in #49. The lib compiles to a framework now! Hurray! 

Carthage build --no-skip-current --platform iOS also works fine:

```
Alexs-iMac:BluenetLibIOS alex$ carthage update --platform iOS --no-use-binaries
*** Fetching IOS-Pods-DFU-Library
*** Fetching SwiftyJSON
*** Fetching PromiseKit
*** Fetching Zip
*** Checking out CryptoSwift at "0.6.1"
*** Checking out PromiseKit at "4.1.3"
*** Checking out SwiftyJSON at "3.1.4"
*** Checking out Zip at "0.6.0"
*** Checking out IOS-Pods-DFU-Library at "3.0.4"
*** xcodebuild output can be found in /var/folders/lv/t0vwd5w9211_mrcjqvc1gcdw0000gn/T/carthage-xcodebuild.aqTdFJ.log
*** Building scheme "CryptoSwift" in CryptoSwift.xcworkspace
*** Building scheme "PromiseKit" in PromiseKit.xcodeproj
*** Building scheme "SwiftyJSON iOS" in SwiftyJSON.xcworkspace
*** Building scheme "Zip" in Zip.xcodeproj
*** Building scheme "iOSDFULibrary" in iOSDFULibrary.xcworkspace
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetLibIOS/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetLibIOS/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetLibIOS/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetLibIOS/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
Alexs-iMac:BluenetLibIOS alex$ 
```

I have my dev app, my consumer app, and thrid party apps which uses the lib through carthage. The only entry in those cart files is the lib git repo:

```
# BluenetLibIOS
# link to the local repo so I don't have to tag releases for experimenting with Carthage.
git "file:///Users/alex/Dropbox/DoBots/Projects/XCode/BluenetLibIOS/.git" "master"

# BluenetLibIOS on github
#github "crownstone/bluenet-lib-ios"
```

If I now run carthage update --platform iOS --no-use-binaries I get this:

```
Alexs-iMac:BluenetRuntime alex$ carthage update --platform iOS --no-use-binaries
*** Fetching 
*** Fetching IOS-Pods-DFU-Library
*** Fetching SwiftyJSON
*** Fetching PromiseKit
*** Fetching Zip
*** Checking out CryptoSwift at "0.6.1"
*** Checking out PromiseKit at "4.1.3"
*** Checking out SwiftyJSON at "3.1.4"
*** Checking out Zip at "0.6.0"
*** Checking out IOS-Pods-DFU-Library at "3.0.4"
*** Checking out  at "6b576bb8e5b2cc7c4f76ac25cac0c0dc877843b9"
*** xcodebuild output can be found in /var/folders/lv/t0vwd5w9211_mrcjqvc1gcdw0000gn/T/carthage-xcodebuild.9fXSgA.log
*** Building scheme "CryptoSwift" in CryptoSwift.xcworkspace
*** Building scheme "PromiseKit" in PromiseKit.xcodeproj
*** Building scheme "SwiftyJSON iOS" in SwiftyJSON.xcworkspace
*** Building scheme "Zip" in Zip.xcodeproj
*** Building scheme "iOSDFULibrary" in iOSDFULibrary.xcworkspace
*** Building scheme "CryptoSwift" in CryptoSwift.xcworkspace
*** Building scheme "SwiftyJSON iOS" in SwiftyJSON.xcworkspace
*** Building scheme "BluenetLibIOS" in BluenetLibIOS.xcodeproj
*** Building scheme "iOSDFULibrary" in iOSDFULibrary.xcworkspace
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
/Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/CryptoSwift/Sources/CryptoSwift/ChaCha20.swift:16:21: warning: property should not be declared fileprivate because its type uses a private type
A shell task (/usr/bin/xcrun xcodebuild -workspace /Users/alex/Dropbox/DoBots/Projects/XCode/BluenetRuntime/Carthage/Checkouts/IOS-Pods-DFU-Library/Example/iOSDFULibrary.xcworkspace -scheme iOSDFULibrary CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES clean -showBuildSettings) failed with exit code 65:
xcodebuild: error: The workspace named "iOSDFULibrary" does not contain a scheme named "iOSDFULibrary". The "-list" option can be used to find the names of the schemes in the workspace.
```

I would assume that the package manager will just build the dependencies as it did before using the iOSDFULibrary.

Regards

mostafaberg, 2017-01-18:
That is true, i'll take a look right now and let you know my findings soon !, thanks for the tip

AlexDM0, 2017-01-18:
On a side note, when I used the following run script phase

```
# required to build framework in xcode
rm -f ./Carthage/Checkouts/Zip/Zip/minizip/module.modulemap
```

I got different errors mentioning minizip. I tried adding minizip to carthage explicitly, but that did not work either.

mostafaberg, 2017-01-18:
Yeah that's understandable, I'm still working on it, builds are extremely slow, I also forked your project in case i need to do any changes, if i find an issue with your configs that i manage to resolve will just send you a PR, if not we will fix iOSDFULibrary and let you know 👍

AlexDM0, 2017-01-18:
Great! I hope this can be resolved soon since we need to push an update to our users that will update their firmware in a few weeks. 

Cheers!

closed 2017-01-18 by mostafaberg

AlexDM0, 2017-01-18:
Thanks a lot!

mostafaberg, 2017-01-20:
Just a quick heads up, 3.0.5 have introduced an issues that might affect your build, she Zip scheme wasn't shared and wasn't built by carthage, however if you already had it in your project, things should work.

but if you were building from scratch carthage wouldn't pick up the zip framework and you'll get a runtime like this: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/78

This has been resolved in 3.0.6 and i would recommend testing that out :)

AlexDM0, 2017-01-20:
Cool Thanks! I'll give it a try!

AlexDM0, 2017-01-20:
Hi, I tried it just now and while the carthage update works, the actual building of the project itself (using the play button) does not work unless I remove the checkout folder.

Project example: 
https://github.com/crownstone/bluenet-example-ios

```
<snippit>/Carthage/Checkouts/IOS-Pods-DFU-Library/Example/Pods/Zip/Zip/minizip/module.modulemap:1:8: Redefinition of module 'minizip'

could not build Objective-C module 'Zip'
```

Ironically carthage build does work. Any ideas other than adding a remove script again?

Regards

mostafaberg, 2017-01-23:
@AlexDM0 That's interesting, do you by any chance have a `Zip.frmework` lingering around in your project ?

mostafaberg, 2017-01-23:
@AlexDM0, I did take a look now, Zip was a submodule still causing the issue you're seeing, that's why the Carthage build would work, but Xcode would notice a redefinition of the Zip module.

i sent you a PR https://github.com/crownstone/bluenet-example-ios/pull/1 give it a shot and let me know if you still encounter issues, if you do i would say try to clear all Carthage related files/directories, and make sure you no longer have any extra references in the project, then do a new bootstrap

AlexDM0, 2017-01-23:
Hi,

While the solution works, it is kind of against Carthage protocol. From their README:

```
The basic workflow looks something like this:

Create a Cartfile that lists the frameworks you’d like to use in your project.

Run Carthage, which fetches and builds each framework you’ve listed.

Drag the built .framework binaries into your application’s Xcode project.

Carthage builds your dependencies and provides you with binary frameworks, but you retain full control over your project structure and setup. Carthage does not automatically modify your project files or your build settings.
```

I don't mean to make an issue of things that essentially work fine, but given that we're providing our lib to others as well, I'd like it to be working according to the normal Carthage workflow. Is this a feasible goal?

Regards

mostafaberg, 2017-01-23:
Maybe I'm misunderstanding something ? this is how the workflow is, my fix was only removing an extra `Zip` module, not the actual one that is exported, I see your `Cartfile` has the correct dependencies, our library has the `Zip` module dependency and does build it as it should, so I'm not sure where this breaks the proper workflow, can you elaborate a bit more ? We do care about having a proper library so I'd like to fix anything that doesn't seem right ofcourse:)

AlexDM0, 2017-01-23:
Hi, sorry, turns out it's something else...

If I just open the xcodeproj file for the example app, it all works. If I have an XCode workspace, for some reason I cannot compile the app. Because this went wrong I misunderstood the changes you provided.

I'll try to further look into the workspace issue. I made a new workspace and added the bluenetioslib xcodeproj and the bluetnet example ios xcodeproj files and then it does not compile.

Regards

mostafaberg, 2017-01-23:
Ah workspace, that's a bit different yes, I didn't do any changes except in the actual files and haven'd changed the XCode project, if you need help with the Workspace problems let's take it in a new issue, might be helpful for someone else in the same situation later :)

commit 21eb02dd (2017-01-18): +Updated release 3.0.5 that fixes #76 and #49