nordicsemi/IOS-DFU-Library #78: dyld: Library not loaded: @rpath/Zip.framework/Zip w/ Carthage

jakerockland, 2017-01-18:
I believe that this was caused by #77

mostafaberg, 2017-01-19:
Taking a look at that now, how do I reproduce it ?

mostafaberg, 2017-01-19:
Cannot reproduce here so I'm not sure if it has something to do with your project's configuration ? can you reproduce on a new empty project ? this will rule out one issue at least.

jakerockland, 2017-01-19:
Definitely can reproduce with an empty project. Here is a dummy project that shows the issue.

[HelloWorld.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/717417/HelloWorld.zip)

jakerockland, 2017-01-19:
#77 removed the `Zip` dependency, preventing it from being build as an additional framework for Carthage users 😢

mostafaberg, 2017-01-20:
@jakerockland Of course !, how silly of me, so sorry for that mistake, i seem to have forgotten to share the Zip scheme so it won't be build by Carthage , working on fix now :)

mostafaberg, 2017-01-20:
Issue resolved in `3.0.6` by making the Zip framework scheme shared. now carthage picks it up and builds both iOSDFULibrary and Zip frameworks!

give it a shot and let me know if you experience any more issues!

closed 2017-01-20 by mostafaberg

jakerockland, 2017-01-23:
I'm still seeing the same problem, same `dyld: Library not loaded: @rpath/Zip.framework/Zip` runtime crash with `3.0.6` using the same example project I had included above.

jakerockland, 2017-01-23:
😱 😅

jakerockland, 2017-01-23:
Whoops, that was actually an error on my end-- `3.0.6` is working now for me! Thank you @mostafaberg 😅 😄