nordicsemi/IOS-DFU-Library #79: Error 15: Invalid Object

kevinpk, 2017-01-19:


philips77, 2017-01-20:
The debug key you are using to test is a public key, but should be a private key. Private key has been deleted just after creating test samples in ble_dfu_send_hex and can't be restored for security reasons (customers would use it instead of generating their own one).
There are two solutions what you can do:
1. Try the zip files from ble_dfu_send_hex folder; they will work if you flash the BL using hex from this folder, or one you compile yourself using the debug public key.
2. Generate a pair of private and public keys yourself and compile BL using the new public key and create a zip from app using the new private key.

kevinpk, 2017-01-20:
Ok thanks @philips77. I didn't realize I wasn't signing both packages with the same key. So where I am now is, following your advice, I generated a debug zip package for the iOS app to send down signed by my own private key , and I am trying to compile a new boot loader with that same private key. I modified 'dfu_public_key.c' with my private key data and ran make, resulting in:

```
fatal error: uECC.h: No such file or directory
 #include "uECC.h"

No logical Google searches have yielded where this uECC header file might be in regards to Nordic's SDK. I'm sorry if this has gone beyond the cocoa pods issue realm. I'll head over to the Nordic dev support site for further help, unless it's something obvious to you! Thanks again.
```

philips77, 2017-01-21:
Hi, could you please post your question on DevZone? We here are not experts in firmware development. Our support team is more capable of helping you. Thanks.
I'll close the issue here as it's not related to the DFU lib.

closed 2017-01-21 by philips77

kevinpk, 2017-01-23:
uECC question answered here on DevZone: https://devzone.nordicsemi.com/question/93540/sdk12-dfu-examples-compilation-error/?answer=93554#post-id-93554