nordicsemi/IOS-DFU-Library #81: Compiling lib containing the DFU lib in xcode workspace does not work

AlexDM0, 2017-01-23:
Hi, 

I'm using XCode 8 and swift 3. If I create a new workspace and add the following projects:

https://github.com/crownstone/bluenet-lib-ios

and

https://github.com/crownstone/bluenet-example-ios

somehow the bluenet example can not build. In this case, the bluenet-lib-ios is the lib that contains the DFU lib.

Any ideas?

mostafaberg, 2017-01-23:
What's the errors that you're getting ?

AlexDM0, 2017-01-23:
Right, sorry, initially, I got the same zip errors but when I created a new workspace I get:

```
Command /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc failed with exit code 1
```

I tried making sure the optimization flags are None but no success so far. If I open the individual projects they do build fine.

mostafaberg, 2017-01-23:
Hmmm, that error is not very descriptive, can you check the buildtime error log and expand that error and see what's happening in there ?
[image]

AlexDM0, 2017-01-23:
Here you go:

```
CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler
    cd /Users/alex/Dropbox/DoBots/Projects/XCode/bluenet-example-ios
    export DEVELOPER_DIR=/Applications/Xcode.app/Contents/Developer
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    export SDKROOT=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS10.2.sdk
    export TOOLCHAINS=com.apple.dt.toolchain.XcodeDefault
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc -incremental -module-name DevStone -Onone -sdk /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS10.2.sdk -target arm64-apple-ios9.0 -g -module-cache-path /Users/alex/Library/Developer/Xcode/DerivedData/ModuleCache -Xfrontend -serialize-debugging-options -embed-bitcode-marker -enable-testing -I /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Products/Debug-iphoneos -F /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Products/Debug-iphoneos -F /Users/alex/Dropbox/DoBots/Projects/XCode/bluenet-example-ios/Carthage/Build/iOS -c -j8 /Users/alex/Dropbox/DoBots/Projects/XCode/bluenet-example-ios/Bluenet/ViewController.swift /Users/alex/Dropbox/DoBots/Projects/XCode/bluenet-example-ios/Bluenet/JSONUtils.swift /Users/alex/Dropbox/DoBots/Projects/XCode/bluenet-example-ios/Bluenet/AppDelegate.swift -output-file-map /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/Objects-normal/arm64/DevStone-OutputFileMap.json -parseable-output -serialize-diagnostics -emit-dependencies -emit-module -emit-module-path /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/Objects-normal/arm64/DevStone.swiftmodule -Xcc -I/Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/swift-overrides.hmap -Xcc -iquote -Xcc /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/DevStone-generated-files.hmap -Xcc -I/Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/DevStone-own-target-headers.hmap -Xcc -I/Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/DevStone-all-non-framework-target-headers.hmap -Xcc -ivfsoverlay -Xcc /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/all-product-headers.yaml -Xcc -iquote -Xcc /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/DevStone-project-headers.hmap -Xcc -I/Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Products/Debug-iphoneos/include -Xcc -I/Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/DerivedSources/arm64 -Xcc -I/Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/DerivedSources -Xcc -DDEBUG=1 -emit-objc-header -emit-objc-header-path /Users/alex/Library/Developer/Xcode/DerivedData/BluenetWorkspace-ctwjboexknfjsaemurnuvcntfmgk/Build/Intermediates/DevStone.build/Debug-iphoneos/DevStone.build/Objects-normal/arm64/DevStone-Swift.h -Xcc -working-directory/Users/alex/Dropbox/DoBots/Projects/XCode/bluenet-example-ios

<unknown>:0: error: merge-module command failed with exit code 1 (use -v to see invocation)
Command /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc failed with exit code 1
```

mostafaberg, 2017-01-23:
Trying it out locally now, what are the steps to recreate the exact workspace as yours ?

AlexDM0, 2017-01-23:
Checkout both projects. Run carthage update --platform iOS --no-use-binaries on both projects via commandline.

in xcode (close any windows if open), new workspace, drag the bluenetlibios.xcodeproj and the devstone.xcodeproj files into the workspace, press the play button for the devstone project (devstone is bluenet-example-ios).

mostafaberg, 2017-01-23:
What about the Cartfile in the `bluenet-example-ios` project, it should point to the `file://xxx.git` loacl repo now instead of the remote git repo right ? or are you still pointing to the repo on Github ?

AlexDM0, 2017-01-23:
I'm still pointing to the one on github. Though during development it often switches. The current issue is while pointing to the git repo

mostafaberg, 2017-01-23:
Sorry for the delays, takes a bit time to build, I'm currently stuck at the redefinition of `minizip` error , seems like something went wrong with the configuration or i had an older version being built, will resume work on that tomorrow!, but I didn't encounter the error you posted yet.

AlexDM0, 2017-01-23:
I had the minizip as well, maybe its the same cause. I could not for the life of me get it to build so I deleted my files and checked out the ones from git. Its tricky and build does take forever! Thanks so much for the help!

mostafaberg, 2017-01-24:
Yeah i'm thinkin it's the same issue since the library is already in the workspace, and we are importing the framework via carthage, so it's basically there twice, may I ask why you need to have the workspace setup ?

AlexDM0, 2017-01-24:
I use the workspace to quickly change the lib and compile the test app for verification. It's not required but it's a nice workflow. The lib itself can't be run of course. 

If need be I can move away from this setup though. What puzzles me though is that this is the only lib that causes this.

mostafaberg, 2017-01-24:
Yeah I get that now, it is not a bad setup in theory and should work, i personally prefer to have a more controlled reproducible environment, so I use a cocoapods development pod workflow for that, gives me a workspace with a target for the library + a target for the example app, but that requires cocoapods.. i'm still looking into your issue now and will update you with my findings

mostafaberg, 2017-01-24:
Seems like my guess was right, the library was imported once as a framework (when you run carthage updat on the example app project), then referenced again when you add the library project to your workspace.

what I did was checkout both projects, add them to a new workspace, then run carthage updat ONLY on the library project, once that is fully working, the example app procect will have it's only dependency available (the library) and no extra work is needed, let me know how it goes.

I'll leave the isuse open until you can confirm it worked for you.

AlexDM0, 2017-01-24:
I'll try to verify this asap. It's odd though, I thought workspaces were just collections of projects. They shouldn't change the options of compiling individual projects. 

Doesn't this imply that the example xcodeproj would not compile anymore when opened individually? 

Cheers

mostafaberg, 2017-01-24:
@AlexDM0 Yes and no, to be able to build the project separately you'll need to update the library procject, push to github, checkout a new version of the example project and pull dependencies from github. Workspaces share the build directory between child projects, and automatically handles the dependencies for you, please take a look at this page that explains in better detail the process of libraries shared between multiple child projects

https://developer.apple.com/library/content/featuredarticles/XcodeConcepts/Concept-Workspace.html

AlexDM0, 2017-01-24:
Thats a shame. Your solution works though but so does removing the checkouts folder manually.. I guess that solution is preferable then since the run script rm -rf /Carthage/Checkouts does not interfere with the rest of the setup.

mostafaberg, 2017-01-24:
You actually don't need to delete any files manually, just running `carthage update` on the library project and not the example app should work flawlessly, you need to delete the checkouts in the example app project ONLY if you have run `carthage update` on it, which is your case, so you'll only need to delete that once. no further modifications needed, will now close this issue since you managed to get it working 👍

closed 2017-01-24 by mostafaberg

AlexDM0, 2017-01-25:
While everything works great in the workspace if I add the rm -rf checkouts step in the build of the example project, I get the missing minizip error if I open just the xcodeproj of the example project and do not have the checkouts folder. Shouldn't the minizip module just be a carthage module?

mostafaberg, 2017-01-26:
You shouldn't need the `rm -rf` anymore, if you still need to I would recommend cleaning your cartahge cache just and refetch in case, you will get the missing minizip error if you just open the `xcodeproj` after deleting the checkouts dir since it's been removed by the build step

AlexDM0, 2017-01-26:
But the minizip is actually living in the checkouts folder right? Shouldn't the checkout folder be completely irrelevant?

mostafaberg, 2017-01-26:
it is, unless you're using submodules then i think it'll be linked, I need to verify my understanding of this setup as I might be wrong, but there is for sure an issue with the `minizip.modulemap` setup that i'm trying to work out, unfortunately `minizip` is bundled directly in the `Zip` framework and is native C so that's why we need a modulemap, Ideally it would be setup as a separate dependency but that's not the case at the moment, so will have to work out something smart until then! sorry for the inconvenience