nordicsemi/IOS-DFU-Library #82: Quick Connect/Disconnect followed with error

drewpitchford, 2017-01-25:
I'm attempting to use this pod in my app. I've followed the instructions from [here](https://raw.githubusercontent.com/NordicSemiconductor/IOS-Pods-DFU-Library/3.0.5/documentation.md) as well as downloaded the open source iOS app. The app will connect and disconnect within a second with the error "DFU Control Point does not have the Write and Notify properties: 302". My firmware guys say they have successfully used the store app to update, so the FW should be good to go. Any helpful pointers to get this figured out? Thanks!

mostafaberg, 2017-01-26:
Well if the project builds and you can start the DFU process i would say your installation was ok, can you download the app store version and see if you can flash it there yourself ?

mostafaberg, 2017-01-26:
The error is pretty descriptive, something is wrong on the peripheral side since the control point characteristic has no write/notify properties, I would say make sure it's on the latest firmware version by any other means, then try to flash that with the library to see how it goes, there are many possibilities on what's wrong, but its peripheral side for sure

mostafaberg, 2017-02-01:
Any updates on this issue ?, will close later and assume it's resolved if i get no reply within a couple of days

closed 2017-02-07 by mostafaberg