nordicsemi/IOS-DFU-Library #83: No null safety for DFUPeripheral

casey-pax, 2017-01-27:
If we pass anything nil through to the DFUPeripheral, there's no safety checking. Then things like the following (line 100) crash
`if peripheral!.state != .connected {`

I think I was passing a nil target, which I have fixed in our code. But it would be nice to get a nil DFUServiceIntiator for example if the target or centralManager were nil.

mostafaberg, 2017-01-30:
Absolutely! thanks for the tip, will get a fix out asap

casey-pax, 2017-03-31:
+1. Have been seeing this in our logs. I do a nil check on the cbPeripheral for setting up the DFUServiceInitiator, but I suppose it's possible the peripheral disconnects in the microseconds between the nil check....

philips77, 2017-04-01:
It doesn't need to be connected. The lib will connect if needed. Jest can't be nil. I'll add check in the next release which is coming next week (support from buttonless service from sdk 13 added).

casey-pax, 2017-04-02:
Thanks, looking forward to the update.

Yes, sorry I meant that the peripheral disconnects at which point my own code nils it out:
`- (void)doUpdate(Foo *bar) {
     if(bar.cbPeripheral == nil) {
        return;
    }
                <-- perhaps another thread nils out bar.cbPeripheral here?
    // Initialize service initiator with bar.cbPeripheral, which is now nil.
}`

closed 2018-10-05 by philips77