nordicsemi/IOS-DFU-Library #84: Inconsistency with Android DFU library – progress speed

tomasbedrich, 2017-01-28:
The progress speed in Android version is in KB/s and in B/s in this version. It would be nice to unify the units.

mostafaberg, 2017-01-30:
Great point!, thanks for your feedback, don't know how this escaped us, will get a fix as soon as we can :)

philips77, 2017-01-30:
That would break compatibility as API would change.

mostafaberg, 2017-01-31:
@philips77 it sure will, but what about having it as a single update, and since we use semantic versioning people who automatically update will still be safe as the library won't update any breaking versions. I'm also thinking we can do that change after we have all other bugfixes & features pushed so people don't have to deal with it if they want to get the latest release?

closed 2018-10-05 by mostafaberg