nordicsemi/IOS-DFU-Library #85: DFU update fails

anlaital, 2017-02-02:
This was previously discussed here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/69

Now I've created a separate test application with a rewritten BT stack and this still happens with our peripheral.

Here are the logs from a successful case running on `3.0.0`:

```
Connecting to 9_DC0C893EF15D...
centralManager.connect(peripheral, options:nil)
Connecting
[Callback] Central Manager did connect peripheral
Connected to 9_DC0C893EF15D
Discovering services...
peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123, FE59])
Services discovered
Connected to 9_DC0C893EF15D
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Starting
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Writing image sizes (0b, 0b, 74700b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0000000000000000cc230100, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Start DFU (Op Code = 1, Upload Mode = 4) request sent
Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
Response (Op Code = 1, Status = 1) received
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Uploading
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Uploading firmware...
Sending firmware DFU Packet characteristic...
```

Here are the logs from a failing firmware update on `3.0.6`:

```
Connecting to 9_DC0C893EF15D...
centralManager.connect(peripheral, options: nil)
Connecting
[Callback] Central Manager did connect peripheral
Connected to 9_DC0C893EF15D
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Legacy DFU...
Connected to 9_DC0C893EF15D
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Starting
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Application with buttonless update found
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Enabling DFU Mode
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
```

Let me know if we need to send you one of our peripherals for investigating this further or if there is any other need for assistance.

mostafaberg, 2017-02-02:
Seems like something is wrong with the buttonless DFU step, is the device in bootloader mode or application mode when you run those tests ? also, can we know a bit more about the softdevice version and the SDK version you're using ? I don't think we need a peripheral, but a distribution packet or the hex files used would help us to reproduce the issue locally!

sorry about the inconvenience

anlaital, 2017-02-02:
nRF51SDK version is 6.1.0 and SoftDevice S110 version is v7.3.0

It's in bootloader mode. As I said, everything works perfectly fine with `3.0.0` release of `iOSDFULibrary`, but anything later just doesn't.

I can get `3.0.6` working with following modifications:

```
    internal var requiredServices: [CBUUID]? {
        // BEGIN HACK
        // Discovering all services (returning nil here) breaks things for some very obscure and unknown reason...
        // This is the change needed to get things working again:
        return [LegacyDFUService.UUID, SecureDFUService.UUID]
        // END HACK

        // We have to find all services, not only those releated to DFU. This is required in case the target device
        // was created using SDK 6.0 or 6.1, where there was no DFU Version characteristic. In that case, this DFU library determines
        // whether to jump to bootloader, or proceed with DFU based on number of services found. We have to find all of them.
        // It is not necessary for newer firmwares (SDK 7+) or for Secure DFU where the code below could work.
        return nil
        
        /*
        // If the experimental feature was enabled
        if experimentalButtonlessServiceInSecureDfuEnabled {
            return [LegacyDFUService.UUID, SecureDFUService.UUID, SecureDFUService.ExperimentalButtonlessDfuUUID]
        }
        // By default only standard Secure and Legacy DFU services will be discovered
        return [LegacyDFUService.UUID, SecureDFUService.UUID]
        */
    }
```

mostafaberg, 2017-02-03:
Perfect, I'll take a look into that issue today and let you know my findings, seems like there's absolutely no reason for this to happen which is strange, but hopefully if i dig deeper i might find something hidden, if it's an issue with the library i'll just push out a fix right away.
thanks! when you have time is it possible to s end us the hex files you're using for SD and BL ?

anlaital, 2017-02-03:
Certainly. Where can I send the files to @mostafaberg?

mostafaberg, 2017-02-03:
you can use mostafa.berg@nordicsemi.no

mostafaberg, 2017-02-07:
haven't gotten any emails yet

anlaital, 2017-02-07:
My apologies for the late response. I've sent the SD and BL hex files to your email now.

mostafaberg, 2017-02-07:
No worries :), got your mail now, will do some tests and let you know my findings

mostafaberg, 2017-02-07:
I've shot you another email, did you receive that ?

jotaefejota, 2017-03-16:
Hi @mostafaberg , I have a very similar issue.

See my logs:

> DFU Log 5: Connected to device
> DFU Log 1: Discovering services...
> DFU Log 0: peripheral.discoverServices(nil)
> DFU Log 5: Services discovered
> DFU Log 1: Starting Legacy DFU...
> DFU Log 5: Connected to device
> DFU Log 5: Services discovered
> DFU Log 1: Legacy DFU Service found
> DFU Log 1: Discovering characteristics in DFU Service...
> DFU Log 0: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
> DFU Log 5: DFU characteristics discovered
> DFU Log 1: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
> DFU Log 1: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: DFU Control Point notifications enabled
> DFU Log 15: Application with buttonless update found
> DFU Log 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.writeValue(0xd7ae, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
> DFU Log 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

(and nothing else happens)

iOSDFULibrary v3.0.6
Bootloader S130 1.0.0
Softdevice S130 1.0.0
Nrf51 S130 1.0.0

Any pointers?
Thanks

mostafaberg, 2017-03-16:
@jotaefejota can you open up a new issue so we can take it there ? thanks

mostafaberg, 2017-03-16:
@anlaital Any news if this worked for you or not ? if i don't hear back within a couple of days I'll asume you resolved it :)

anlaital, 2017-03-16:
Sorry I haven't gotten back to you. Completely forgot in the hurry that we've been having. :)

This is still an issue and occurs in a stand-alone tester app. @jotaefejota has the exact same issue as we do.

mostafaberg, 2017-03-17:
@anlaital you say that @jotaefejota has the exact same issue, he triggers that by manually disconnecting during the DFU process, do you do the same ? or does it always happen ? if it doesn't then it's not an identical issue

mostafaberg, 2017-03-17:
@philips77 IIRC was that something you fixed before ?

philips77, 2017-03-17:
Hi, this seems to be easy. On sdk 6.1 there was no DFU Version characteristic, based on which in newer versions I determine whether phone has connected to the app (value returned os 0x0100) or to the bootloader (versions 0x0K00, K ={5,6,8}).
If this characteristic is not found, that means that it's sdk 6.1, which supported app update and buttonless jump. I have no way to find out that I'm connected to, so i count services. If only DFU is found, I assume it's a bootloader, if more - an app with buttonless jump feature.
In this case in 3.0.0 the lib was searching only for dfu service, so was always assuming DFU bootloader mode. This bug was fixed later, and I guess this causes troubles in your case. However, there are implementations where this assumption is not true, for example when the dfu process can be done in app mode without jumping. Then set https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L130 to true.

anlaital, 2017-03-17:
The change that triggers this behaviour is https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/compare/3.0.0...3.0.1#diff-b906bc607dacb47dda5a8f7ff1a77068L64 

This issue doesn't occur on `3.0.0` version of the SDK. Every version after that is affected.

anlaital, 2017-07-26:
Closing this as a workaround was found. Setting `forceDFU` to `true` resolved the issue.

closed 2017-07-26 by anlaital