nordicsemi/IOS-DFU-Library #87: Delegate example in Obj-C?

benjamincarp, 2017-02-06:
In the sample code in the readme it shows self is the delegate, logger, and progeressDelegate:
```
initiator.logger = self; // - to get log info
initiator.delegate = self; // - to be informed about current state and errors 
initiator.progressDelegate = self; // - to show progress bar
```

Since the example app https://github.com/NordicSemiconductor/IOS-nRF-Toolbox has been moved to swift, can you provide and example of how to handle these duties in Obj-C?

mostafaberg, 2017-02-06:
Ofcourse, pretty simple:
first after installing the pod, navigate te ObjC class you'll be using the library in, at the top insert:

`@import iOSDFULibrary;`

and then in any method where you'll start the DFU process, you can do something like that:
```
DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:myCentralManager target:myTargetPeripheral];
[initiator setLogger:self];
[initiator setDelegate:self];
[initiator setPacketReceiptNotificationParameter:12];
[initiator setProgressDelegate:self];
etc ...
```

closed 2017-02-06 by mostafaberg

benjamincarp, 2017-02-06:
I got that far. What I was asking was for an example of actually handling the logging, delegation, progress that this library is going to report. 

I assume there is an interface I am supposed to implement that will supply functions to be called for each of these things (an on progress update handler for instance). But I'm not finding any documentation on that step. 

I dug through old commits of the example and found where it had implemented `DFUViewController` but that doesn't appear to be part of the framework anymore (or I'm not pulling it in correctly) as I get errors if I try to do the same.

mostafaberg, 2017-02-06:
You don't need to manually pull the framework, you just need to import it using `CocoaPods`

Those delegates and their protocols are as follows:
For the initiator delegate and progress delegate, implement `DFUServiceDelegate` and `DFUProgressDelegate` respectively, their implementation can be found [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUServiceDelegate.swift)

For the logger delegate, implement the protocol `LoggerDelegate` found [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Utilities/Logging/LoggerDelegate.swift)

Please note as the Obj-C code isn't in the library, Xcode generates that for you after you install the pod.

We no longer officially support Obj-c and are moving forward with only supporting Swift 3 and above, however Obj-C is fully functional, but we will not provide docs for that as it's not an officially supported feature.

When you add any of the protocols to your classes, just Command+Click on the protocol name to see the Obj-C implementation that Xcode has generated for you.

If you require further assistance setting up an Obj-C Project, we recommend taking this to our [DevZone](https://devzone.nordicsemi.com) as you will get much more response there as we only use Github to track issues

benjamincarp, 2017-02-06:
Thanks. That is what I was looking for. I'll take further questions over to the devzone.