nordicsemi/IOS-DFU-Library #88: Use of undefined identifier 'DFUStateSignatureMismatch' in Objective C

nhatminh12369, 2017-02-07:
My project is written in Objective C and I am trying to use this library. However, this is the error that I am getting:
`Use of undefined identifier 'DFUStateSignatureMismatch'`

I have tried to install this library into my project using Cocoapod, Carthage and also copy the code directly into my project but I am still getting that error. Please help!

mostafaberg, 2017-02-07:
We do not have any enum named DFUStateSignatureMismatch in the library, you probably updated the iOSDFULibrary with a breaking changes version and didn't update your code to follow those changes, can you show us some code snippet where this is triggered ?

mostafaberg, 2017-02-08:
@nhatminh12369 Did you manage to resolve your issue ?

nhatminh12369, 2017-02-08:
I have resolved the issue above. I was because I upgraded the library to the latest version and didn't notice that this enum is no longer exists: DFUStateSignatureMismatch.
Thanks @mostafaberg.

closed 2017-02-08 by mostafaberg