nordicsemi/IOS-DFU-Library #89: Error while trying to assign zip file url to selected firmware

aahirgiri, 2017-02-07:
This is my URL: file:///Users/aahir/Desktop/DFU/DFU%20Test/yoky_dfu.zip

DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:url];

But after this step, the above variable is still set to nil. Hence it is crashing after [initator start];

*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[ViewController dfuError:didOccurWithMessage:]: unrecognized selector sent to instance 0x12dd09d60'
*** First throw call stack:
(0x18f8851b8 0x18e2bc55c 0x18f88c268 0x18f889270 0x18f78280c 0x1001f26d4 0x1001f282c 0x100072818 0x10006fa44 0x100b6d258 0x100b6d218 0x100b7b26c 0x100b7ce2c 0x100b7cb78 0x18e9172a0 0x18e916d8c)
libc++abi.dylib: terminating with uncaught exception of type NSException

This is the error. Am I sending the right URL or should I attach it elsewhere within the app and not from the local directory?

mostafaberg, 2017-02-07:
Yeah it has to be bundled within the app and not on your computer, you can see an example here:

https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/Example/iOSDFULibrary/DFUViewController.swift#L92

or simply in your terminal type `pod try iOSDFULibrary` to see it in action

aahirgiri, 2017-02-07:
Works! Thanks a lot

closed 2017-02-07 by aahirgiri

mostafaberg, 2017-02-07:
No worries ! :)