nordicsemi/IOS-DFU-Library #90: dfuStateDidChangeTo: unrecognized selector sent

aahirgiri, 2017-02-07:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[ViewController dfuStateDidChangeTo:]: unrecognized selector sent to instance 0x155e08870'
***
 First throw call stack:
(0x18f8851b8 0x18e2bc55c 0x18f88c268 0x18f889270 0x18f78280c 0x10023f544 0x10022f0d4 0x100bc5258 0x100bc5218 0x100bca280 0x18f832810 0x18f8303fc 0x18f75e2b8 0x191212198 0x1957a57fc 0x1957a0534 0x1000bfb10 0x18e7415b8)
libc++abi.dylib: terminating with uncaught exception of type NSException

I'm calling [initiator start] after assigning zip file and also verified that the selected peripheral and selected firmware have correct values. But I'm unable to figure out why this is happening.

mostafaberg, 2017-02-07:
Probably you're not setting the view controller as the delegate, you may use [this as a reference](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/Example/iOSDFULibrary/DFUViewController.swift#L27)

aahirgiri, 2017-02-07:
I have set the view controller as the delegate. No luck still

mostafaberg, 2017-02-07:
I'm not sure it's hard to tell the issue without seeing the source code, but i would recommend just trying to uninstall the library by doing a `pod deintegrate`, then running `pod update` and try again.

mostafaberg, 2017-02-08:
@aahirgiri Any luck with this yet ?

aahirgiri, 2017-02-08:
No luck. I'm still trying :(

mostafaberg, 2017-02-08:
If you can share the project that causes the issue or the github repo we might be able to take a look and send back a fix

aahirgiri, 2017-02-08:
Thanks. You can find the project here:

https://github.com/aahirgiri/TestProj_DFU.git

aahirgiri, 2017-02-08:
Please pull the latest changes. I have modified the UUID again

mostafaberg, 2017-02-08:
will do 👍

mostafaberg, 2017-02-08:
@aahirgiri Sent you a PR that will hopefully get this working, you forgot to implement the actual delegate methods but defined them in the view's header, I've added some `HACK` warnings in the `.m` file so you just check your warnings and you'll see what needs to be implemented

mostafaberg, 2017-02-08:
@aahirgiri Also remember to only start scanning AFTER the CentralManager's state is ready, before that you should disable the scan button or something since it'll confuse the user when they press scan and nothing happens.

aahirgiri, 2017-02-09:
Works! Thanks a ton :)

closed 2017-02-09 by aahirgiri