nordicsemi/IOS-DFU-Library #91: Using iOS-DFU-Library in Objective-C project

samueleforconi, 2017-02-17:
Hi,
I'm trying to add the iOS-DFU-Library into my application that is written in Obj-C.
To make a test, I created an Obj-C sample project (XCode 8.2.1) and installed the library using the `pod install` command.
When I open the project with `open TestProject.xcworkspace`, XCode tries to convert the sources to Swift 3 syntax and when completed, more than 160 errors comes up... is there something I'm doing wrong in the import procedure?

mostafaberg, 2017-02-17:
which version of cocoapods are you using ?

samueleforconi, 2017-02-17:
```
$ gem which cocoapods
/Library/Ruby/Gems/2.0.0/gems/cocoapods-1.2.0/lib/cocoapods.rb
```

mostafaberg, 2017-03-08:
Can you try installing the --pre version and see is it helps ? Our code is is Swift 3 and there's nothing that needs to be migrated, what is the contents of your podfile ?

samueleforconi, 2017-03-08:
Hi,
it tooks some time, but I managed to make it work... 

1. after installing the pod library, when I first open the project (using `open project.xcworkspace`) I had to stop XCode to try to reformat the code to Swift3 (click on `Later...` button).

2. now, if I try to import the `"iOSDFULibrary-Swift.h"` the header is not found, so I have to change some settings in the configuration window:
- `Other C Flags` and `Other C++ Flags` set to `$(inherited)`
- `Use Legacy Swift Language Version` set to `NO` (for iOSDFULibrary and Zip)

mostafaberg, 2017-03-08:
Ok great to hear it worked, will check out if this is indeed an issue or just a mishap, if it's a bug will resolve it and close the issue, if it's not reproducible i'll let you know and just close the issue.

samueleforconi, 2017-03-08:
Hi, 
the procedure that led me to have errors with the library is the following:

1. create an XCode `Objective-C` project, I've called `TestDFULibrary`

2. create a Podfile, mine was like this:
`platform :ios, '8.0'`
`target 'TestDFULibrary' do`
`use_frameworks!`
`  pod 'iOSDFULibrary'`
`end`

3. install the library with `$ pod install`

4. open the project with `$ open TestDFULibrary.xcworkspace`

5. when the project opens in XCode a window with this message will pop-up:

> Convert to the Current Swift Syntax
The targets "iOSDFULibrary" and "Zip" contain source code developed with an earlier version of Swift.
Choose "Convert" to update the source code in this project to the latest SDKs. You will be given the choice to use Swift 2.3 syntax or update to Swift 3.
This action can be performed later using "Convert to Current Swift Syntax" in the Edit menu.

If you choose "Upgrade", XCode opens a pop-up window 

> Generating preview...

And when it finishes more than 150 errors are present in the project...

mostafaberg, 2017-03-09:
I'll keep an eye out, thanks a ton for the detailed instructions !, i'll let you know my findings when i start working on that

mostafaberg, 2017-03-17:
Seems like things need some ironing with ObjC / Swift and cocoapods
for the time being the best way is to document how to install it for Obj-C separately.

I've created the Obj-C documentation that you can [read here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/README_OBJC.md)

Thanks for your help

closed 2017-03-17 by mostafaberg