nordicsemi/IOS-DFU-Library #92: Invalid Argument Exception

PetrykowskiM, 2017-02-19:
Hello,

I integrated your library into my project in order to implement the DFU upgrade. 

But as it comes to start the upgrade, I get a `NSInvalidArgumentException', reason: '-[BleManager dfuStateDidChangeTo:]: unrecognized selector sent to instance 0x1740d7840'`.

It occurs within the async part in following code from ```DFUServiceSelector```

```
func start() {
        DispatchQueue.main.async(execute: {
           self.delegate?.dfuStateDidChange(to: .connecting)
        })
        peripheral.start()
    }
```

That is the code I am calling it from. Did I do anything wrong? I checked the peripheral and it looked fine.

[image]

Thanks!

mostafaberg, 2017-02-20:
Hi @PetrykowskiM !, I see the address in the error message `0x1740d7840` is pointing to a `CBCentralManager` object in the screenshot. `CBCentralManager` doesn't implement the delegate methods for the DFU service controller, try which class is that code running in ? `BleManager` I would assume ? if so try setting that as the delegate, and make sure it implements the correct delegate methods like `dfuStateDidChangeTo`, etc ...

PetrykowskiM, 2017-03-08:
Thank you, this solved my problem :) Kind of tricky to implement your swift classes with objective c. Maybe you would want to add some documentation for it?

mostafaberg, 2017-03-08:
Glad it worked out, we actually don't support Objective-C that's why there's no documentation for it, Swift 3 is our main focus right now

closed 2017-03-08 by mostafaberg