nordicsemi/IOS-DFU-Library #93: dyld: Library not loaded: @rpath/Zip.framework/Zip

pjl1234567, 2017-03-01:
My Enviorments:
macOS 10.12.2
Xcode 8.2
Library
github "NordicSemiconductor/IOS-Pods-DFU-Library" ~> 3.0.6
github "marmelroy/Zip" ~> 0.6

1. I installed carthage
2.Create a file named cartfile on my computer
3.add the following content to the file:
```
github "NordicSemiconductor/IOS-Pods-DFU-Library" ~> 3.0.6
github "marmelroy/Zip" ~> 0.6
```
1.then I Open a new terminal and cd to the directory where the cartfile  file is
2.Enter the command carthage update --platform iOS
3.last i drag the ```Carthage/Build/iOS/iOSDFULibrary.framework```  and    ```Carthage/Build/iOS/Zip.framework```  to my  project.

but when I run it  follow  the error：
[image]
@mostafaberg

mostafaberg, 2017-03-01:
I think dragging will just add the files to the project and not link them, try adding them from the project configuration's `Link Binary With Libraries` tab if it's not already there

pjl1234567, 2017-03-02:
```Link Binary With Libraries``` tab 
[image]

pjl1234567, 2017-03-02:
[here](https://github.com/pjl1234567/MYDFUTest) is my test code

pjl1234567, 2017-03-03:
@mostafaberg

mostafaberg, 2017-03-03:
Thaks for posting code, will look into it now !

mostafaberg, 2017-03-03:
The libraries aren't being embedded that's why the images can't be found.
You need to add the libraries to the `embedded binaries` tab, you also need to set `always embed swift standard binaries` to `Yes` in your build settings.

closed 2017-03-03 by mostafaberg

pjl1234567, 2017-03-04:
Thanks for your help very much。 @mostafaberg