nordicsemi/IOS-DFU-Library #94: EXC_BAD_ACCESS on the Initiator .with(firmware: ) call.

AccentSystems, 2017-03-09:
Hello. I already commented on the only issue I could find related to this problem but I don't know if anyone will see it as it is already closed.

Today I updated the iOS DFU Library and I am getting the EXC_BAD_ACCESS (code=1, address=0x10). Before the update everything was working properly.

I also tried removing and reinstalling all the pods from scratch.

Here is my code:
```
func setFilePath(){
        let bundleURL = Bundle.main.bundleURL
        let dataFolderURL = bundleURL.appendingPathComponent("firmware")
        let fileURL = dataFolderURL.appendingPathComponent("edd_new.hex")
        
        let fileManager = FileManager.default
        if fileManager.fileExists(atPath: fileURL.path) {
            print("FILE AVAILABLE")
        } else {
            print("FILE NOT AVAILABLE")
        }
        
        let selectedFirmware = DFUFirmware(urlToBinOrHexFile: fileURL, urlToDatFile: nil, type: DFUFirmwareType.application)
        
        startUpdate(selectedFirmware!)
    }
    
    
    func startUpdate(_ fw: DFUFirmware){
        
        if(self.dfuPeripheral == nil){
            print("DFU PERIPHERAL NIL")
        }
        
        let initiator = DFUServiceInitiator(centralManager: centralManager!, target: self.dfuPeripheral!)

        initiator.logger = self; // - to get log info
        initiator.delegate = self; // - to be informed about current state and errors
        initiator.progressDelegate = self; // - to show progress bar
        
        self.dfuController = initiator.with(firmware: fw).start()
    }
```

File is available, and peripheral is not nil. It also crashes on the .with(firmware: fw) part. Tried both to put in on the initiator declaration and before the start as the code shows. Same problem. 

Is there any way I can fix this? Thank you!

mostafaberg, 2017-03-09:
Hi ! can you breakpoint right after the line `let selectedFirmware` and print out the contents of `selectedFirmware` ?

AccentSystems, 2017-03-09:
Hi @mostafaberg , thanks for the fast reply. This is what you need?

[image]

AccentSystems, 2017-03-10:
Hi @mostafaberg. Any update on the issue? I would really appreciate your help as the APP is published with that bug and we need to update it as fast as possible. Our clients are not able to update their devices. Thanks a lot!

AccentSystems, 2017-03-10:
Hi. I was able to fix it by downgrading to pod version 2.0.1. The 3.0.0 also not working. At the moment we will release with this version, but a solution for the newer one would be much appreciated. Thanks.

mostafaberg, 2017-03-13:
Sorry for the delayed response, I will look into it and update you within the week !, glad to hear you have a working version out for now!

mostafaberg, 2017-04-28:
I have checked this over and over and I couldn't reproduce that issue, is it possible that you can make a simple project that reproduces this so i can experiment it ?

AccentSystems, 2017-04-28:
Hi @mostafaberg. I am sorry but I am really busy those days with internal projects. But I promise as soon as I have some time I will do it and let you know. Thank you and BR.

mostafaberg, 2017-04-28:
Ofcourse !, no worries, if i manage to reproduce this i'll let you know here the version that fixes that:) thanks and appreciate the feedback

bisikli, 2017-07-07:
Hello, i think i have the same problem:

[image]

bisikli, 2017-07-31:
Hello @mostafaberg, you had any chance to look into this issue? Thanks

mostafaberg, 2017-07-31:
@bisikli This seems to be a different issue, you should open a separate issue so we can take it there :)

closed 2017-09-22 by mostafaberg