nordicsemi/IOS-DFU-Library #95: Object-C   Undefined symbols for architecture arm64:

xRuby, 2017-03-10:
Undefined symbols for architecture arm64:
  "_OBJC_CLASS_$__TtC13iOSDFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in BluetoothManager.o
  "_OBJC_CLASS_$__TtC13iOSDFULibrary11DFUFirmware", referenced from:
      objc-class-ref in BluetoothManager.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

info: Object-C   XCode8.1

how to resovle....

xRuby, 2017-03-13:
I sovle it  by setup：

ONE： Build Settings －>  Linking  －>  Other Linker Flags  －> $(inherited)
TWO:   Build Settings －>  Build Options  －>  Always Embed Swift Standard Libraries －> Yes

at last , there is no error.     is it right ?

closed 2017-03-16 by xRuby

mostafaberg, 2017-03-17:
Just saw your comment, [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/README_OBJC.md) is our new official instructions for Obj-C Projects!

zhangferry, 2019-07-13:
@mostafaberg I meet the same questions, can you update the documents of Installation instructions for Obj-C projects.