nordicsemi/IOS-DFU-Library #96: Upload Stuck on 'Starting'

andy-fueled, 2017-03-14:
The library works fine with my BT device normally. However, when I disconnect with the device during the upload and try again, I get stuck in the 'Starting' state. Do you know what I am doing wrong? 
Any help would be much appreciated.

Here are the logs from the library (including where I disconnect from the device):
`LOG level: I || message: Connected to DfuTarg
LOG level: V || message: Discovering services...
LOG level: D || message: peripheral.discoverServices(nil)
state changed to: Connecting
LOG level: I || message: Services discovered
LOG level: V || message: Starting Legacy DFU...
LOG level: I || message: Connected to DfuTarg
LOG level: I || message: Services discovered
LOG level: V || message: Legacy DFU Service found
LOG level: V || message: Discovering characteristics in DFU Service...
LOG level: D || message: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
LOG level: I || message: DFU characteristics discovered
LOG level: V || message: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
state changed to: Starting
LOG level: V || message: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: DFU Control Point notifications enabled
LOG level: V || message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
LOG level: V || message: Writing image sizes (0b, 0b, 47500b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x00000000000000008cb90000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
LOG level: I || message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: Start DFU (Op Code = 1, Upload Mode = 4) request sent
LOG level: I || message: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100103
LOG level: A || message: Response (Op Code = 1, Status = 3) received
LOG level: E || message: Error 3: Operation not supported
LOG level: W || message: DFU target does not support DFU v.2
LOG level: V || message: Switching to DFU v.1
LOG level: V || message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x01, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
LOG level: V || message: Writing image size (47500b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x8cb90000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
LOG level: I || message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: Start DFU (Op Code = 1) request sent
LOG level: I || message: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
LOG level: A || message: Response (Op Code = 1, Status = 1) received
LOG level: V || message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
state changed to: Uploading
LOG level: I || message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
LOG level: V || message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
LOG level: I || message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: Uploading firmware...
LOG level: V || message: Sending firmware DFU Packet characteristic...
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 0
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 1
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 2
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 3
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 4
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 5
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 6
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 7
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 8
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 9
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 10
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 11
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 12
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 13
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 14
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 15
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 16
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 17
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 18
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 19
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 20
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 21
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 22
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 23
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 24
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 25
PROGRESS CHANGE : Part: 1. totalParts: 1 progress: 26
LOG level: D || message: [Callback] Central Manager did disconnect peripheral
LOG level: I || message: Disconnected by the remote device
Error: DFUError. message: Device disconnected unexpectedly

------- I disconnect from the device here and restart the upload process ---------------

LOG level: I || message: Connected to DfuTarg
LOG level: V || message: Discovering services...
LOG level: D || message: peripheral.discoverServices(nil)
state changed to: Connecting
LOG level: I || message: Services discovered
LOG level: V || message: Starting Legacy DFU...
LOG level: I || message: Connected to DfuTarg
LOG level: I || message: Services discovered
LOG level: V || message: Legacy DFU Service found
LOG level: V || message: Discovering characteristics in DFU Service...
LOG level: D || message: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
LOG level: I || message: DFU characteristics discovered
LOG level: V || message: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
state changed to: Starting
LOG level: V || message: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: DFU Control Point notifications enabled
LOG level: V || message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
LOG level: V || message: Writing image sizes (0b, 0b, 47500b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x00000000000000008cb90000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
LOG level: I || message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: Start DFU (Op Code = 1, Upload Mode = 4) request sent
LOG level: I || message: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100103
LOG level: A || message: Response (Op Code = 1, Status = 3) received
LOG level: E || message: Error 3: Operation not supported
LOG level: W || message: DFU target does not support DFU v.2
LOG level: V || message: Switching to DFU v.1
LOG level: V || message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x01, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
LOG level: V || message: Writing image size (47500b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
LOG level: D || message: peripheral.writeValue(0x8cb90000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
LOG level: I || message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
LOG level: A || message: Start DFU (Op Code = 1) request sent`

Here is the code I use to reconnect to the device: 
`let aborted = self.firmwareUpdateCoordinator?.controller?.abort()
		if aborted == false {
			LogError("Aborting DFU process failed")
		}
if let peripheral = ConnectionManager.sharedInstance.connectedPeripherals.first,
	peripheral.name == "DfuTarg" {
			let initiator = DFUServiceInitiator(centralManager: manager, target: peripheral).with(firmware: self.latestFirmware!.firmware)
			initiator.logger = self
			initiator.delegate = self
			initiator.progressDelegate = self

			self.controller = initiator.start()
		}

mostafaberg, 2017-03-17:
Let me go through your logs and come back to you in a moment

mostafaberg, 2017-03-17:
How do you disconnect the peripheral ? I would guess you are disconnecting from the phone and the peripheral is still in Bootloader mode and you need to send reset.

It's hard to tell exactly what went wrong without knowing a bit more details about how you disconnect, if you want to see how the nRF Toolbox app handles that you can check its [source code](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox)

andy-fueled, 2017-03-17:
Thank you for looking into this. 

I am disconnecting by putting the BT device into an aluminum box, so the peripheral is still in Bootloader mode. 

What exactly do you mean by 'send reset'? Is that a function from this library or is that something the BT device's firmware should be calling? Does this library's 'abort' function reset the device? 

Sorry if my questions are simple. I am an iOS developer that's working with Bluetooth, and not the other way around.

philips77, 2017-03-17:
Hi @andy-fueled, let me test your case and come back to you in a sec. It's either that your device does not send the 0x100101 notification then, or there is a bug in the library.

We thought you disconnect gracefully, not using a faraday cage :) In that case retrying should succeed without problems and let's see why it doesn't. Will come to you in a moment.

philips77, 2017-03-17:
Does it happen every time you try to repeat like this? Could you send us your nRF51 application and file you want to send using DFU?

philips77, 2017-03-17:
I see one bug in the lib. [Here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L247) should be similar error handler as [few lines above](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L219) as the old DFU also supported resuming so it should return Invalid State after you sent Start DFU (Op Code = 1).

However, this does not explain why didn't you get any notification there. If would be logged had it came.

philips77, 2017-03-17:
Ok, I did some testing by connecting an iPhone to an Android phone with nRF Connect on it that emulates your device (I used a dummy bin file) and if I send any notification there at the end the DFU lib handles it correctly. I tried with 0x100101 - if it was accepted, 0x100102 - what in fact would be returned, as device is in invalid state and 0x100106 - operation failed. All were working correctly despite the bug I mentioned 1 comment above. 

Could you check if you are sending any notification there? Any logs from nRF or a sniffer?

andy-fueled, 2017-03-17:
@philips77 This happens every time. 
I need to get permissions before sharing any code. I will send the files as soon as I am cleared to do so (sorry about the red tape). 

My process to perform the DFU is: 
1. While the BT device is in application mode, the app sends a command to the device that puts the device into DFU mode.
2. The app scans for and connects to `DFUTarg`, and uses that connected peripheral as the target for this library's `initiator` and then I call `start` on the `initiator`.
(As stated in my original message, this process works, unless the upload is interrupted.)

Do you still need the `nRF51 application` code if the DFU is starting when the device is already in DFU mode?

philips77, 2017-03-17:
I don't need the code, just hex files :)

philips77, 2017-03-17:
Is your nRF51 app forked from SDK 6.1 or before? Only then the DFUv1 had to be used.
I have a sniffer here so I could check if the notification has been sent.
Regarding my last comment - I would really prefer hex files only as I don't want to compile anything and see any secret codes :) And no iOS app at all, I think I can use the Example app from this project and modify it a bit if needed?

andy-fueled, 2017-03-21:
@philips77 I made some progress on this issue. 
After my BT device gets disconnected during the upload (by being put in a faraday cage), and the app tries to re-upload the firmware, this happens:

Step 1) `func sendStartDfu` (line 238 of `LegacyDFUService.swift`) is called. The `aborted` bool is false, so a request is sent to the device.
Step 2) The program should continue once the app gets a callback from the device (line 375 of DFUControlPoint.swift). However, my device does not send back a `peripheralDidUpdateValueForCharacteristic` callback, making the app stuck waiting for the callback.

What happens in the NRF-Toolbox when I go through my sequence of steps (described in my original message), is that the program somehow recognizes that the process was aborted, setting the `aborted` bool of `Step 1` (above) to true. That way, the next time I try to upload the firmware, the function in `Step 1` does not send a request, but sends `sendReset(onError: report)`. After that, the device is receptive to re-uploading the firmware.

Right now I can think of 3 possible solutions/workarounds:
1) I can implement a timeout on this library, so that if the device is in the `Starting` state for too long, the app can execute `sendReset(onError: report)`, then re-upload the firmware. However, is the `sendReset(onError: report)` accessible from outside this library?
2) Same timeout, but abort the upload. The problem with this is that whenever I abort and reconnect to restart the upload process, the `abort` variable in `LegacyDFUService.swift` is still always false.
3) The library implements a timeout if there is no callback from the `peripheralDidUpdateValueForCharacteristic`, and executes `sendReset(onError: report)`. 

What do you think?

philips77, 2017-03-21:
Hi Andy,
The ```aborted``` flag is only set when you call ```abort()``` method on the controller. It will not be set when you restart the process. Also, it does not discover that the process was aborted on its own.

Since we talked last time I made 2 commits to develop branch: [this](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/commit/8abe3bed2bfa91e8668e949158d16e149d9d02fa) and [this](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/commit/aca5b7a179fb29684b8e3a7536736eb8d7cd3c81). The first one, as I said, ensures retrying DFU when Invalid State is received from a device running bootloader from SDK before 6.0 (only App update allowed). The other one adds some delay that I found out yesterday that is needed in the iOS DFU Lib for bootloaders before 7.0 (DFU Version char added, Extended Init Packet required). Both commits impact your case, but will not fix the issue you encounter.

In my opinion the issue is in your bootloader implementation. You should check why it doesn't send any notification at the end.

Every other solution would be only a workaround. I think that calling ```abort()``` being in a state when you wait for a notification would not make any difference... This method just sets a ```aborted``` flag that is read in some moments of the process, but not when it actually waits for a notification. The ```sendReset()``` method is not exposed like you said. I'll think about adding a method *forceReset()* if we won't manage to fix your issue by just fixing the root cause.

The last proposal (3.) have some sense. Otherwise, a device that for some strange reason didn't get a notification stays there for ever, and remember that the notifications don't have ACK on the app layer, so the bootloader is not aware of the other side not receiving such. But that would rather implicate changing notifications to indications in Legacy and Secure DFU if that was a case, like they did in the buttonless dfu service in SDK 13. I'll think about it. The problem I see is that in SDK 7-11 the 100101 notification is sent when the flash on nRF is ready for receiving data, so during SoftDevice update it takes ~5-10 sec.

Here's what you should observe in your second connection:
0. First upload terminates with link loss
1. You reconnect to the device. It's now in "Resume upload" state.
2. Lib sends 0x0104 and firmware sizes using DFUv2 procedure which ends with 0x100103 (Not Supported).
3. Lib sends 0x01 and the app size using DFUv1 procedure.
--- here it holds in your case ---
4. The device should respond 0x100102 (Invalid State)
5. The lib should send a Reset command (0x06) and wait until the device disconnects.
6. The lib should reconnect and do 1. , 2. , 3. again
7. Instead of 4. the response now should be 0x100101 (Success)
8. New: Waits for 1 second
9. Sends 0x03 (Receive firmware) command, and so on...

Perhaps the bootloader you have have the resume procedure removed, but is still in the Invalid State which wasn't removed together? And instead of sending 0x100101 sends nothing?

Could you, after you do the first link loss, connect using nRF Connect on Android and try to send the data as your app is doing and see whether you get the notification there? You may record a macro to do this quickly as timing may be crucial. You could post logs here from the app.

andy-fueled, 2017-03-21:
I created the following macro for nRF Connect on Android, but the macro cannot be started and has the following message: `Ensure DFU Packet has WRITE property failed`
This is the macro: 
```<macro name="DFU Steps" icon="DOWNLOAD">
   <assert-service description="Please, connect to a device with Device Firmware Update Service." uuid="00001530-1212-efde-1523-785feabcd123">
   </assert-service>
   <write description="send 0104" service-uuid="00001530-1212-efde-1523-785feabcd123" characteristic-uuid="00001531-1212-efde-1523-785feabcd123" value="0104"/>
   <write description="description" service-uuid="00001530-1212-efde-1523-785feabcd123" characteristic-uuid="00001532-1212-efde-1523-785feabcd123" value="00000000000000008cb90000"/>
   <sleep description="now the macro will wait 1 second..." timeout="1000" />
   <write description="send 01" service-uuid="00001530-1212-efde-1523-785feabcd123" characteristic-uuid="00001531-1212-efde-1523-785feabcd123" value="01"/>
   <write description="send 01" service-uuid="00001530-1212-efde-1523-785feabcd123" characteristic-uuid="00001532-1212-efde-1523-785feabcd123" value="8cb90000"/>
</macro>```

Am I doing something wrong? Is this useful information?

philips77, 2017-03-22:
The write operations that write to DFU Packet must have type="WRITE_COMMAND" attribute, as written here: https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/master/documentation/Macros#write-characteristic
The easiest way to create a macro would just be to record it. Here's what I recorded:
[Dfu bootloader 5.2.xml.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/861591/Dfu.bootloader.5.2.xml.zip)
At the end I assert value 0x100102 = Invalid State, but could be changed to 0x100101 = Success

andy-fueled, 2017-03-22:
Thanks for that xml file. I used that same macro, and nRF Connect never received the last notification from the device. Here are the logs:
nRF Connect, 2017-03-22
DfuTarg (F0:87:E3:3E:D4:C7)
V	10:32:36.554	Connecting to F0:87:E3:3E:D4:C7...
D	10:32:36.555	gatt = device.connectGatt(autoConnect = false)
D	10:32:37.053	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	10:32:37.053	Connected to F0:87:E3:3E:D4:C7
D	10:32:37.053	wait(600ms)
D	10:32:37.064	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	10:32:37.669	Discovering services...
D	10:32:37.669	gatt.discoverServices()
D	10:32:38.152	[Callback] Services discovered with status: 0
I	10:32:38.153	Services discovered
V	10:32:38.168	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
V	10:32:52.073	Writing to descriptor 00002902-0000-1000-8000-00805f9b34fb
D	10:32:52.073	gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0100)
I	10:32:52.104	Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
A	10:32:52.104	"Notifications enabled" sent
D	10:32:52.110	gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
V	10:32:52.112	Notifications enabled for 00001531-1212-efde-1523-785feabcd123
V	10:32:52.112	Writing request to characteristic 00001531-1212-efde-1523-785feabcd123
D	10:32:52.112	gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123, value=0x0104)
V	10:32:52.147	Writing command to characteristic 00001532-1212-efde-1523-785feabcd123
D	10:32:52.147	gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123, value=0x00000000000000008CB90000)
I	10:32:52.148	Data written to 00001531-1212-efde-1523-785feabcd123, value: (0x) 01-04
A	10:32:52.148	"Start DFU
Upload mode: APPLICATION" sent
I	10:32:52.149	Data written to 00001532-1212-efde-1523-785feabcd123, value: (0x) 00-00-00-00-00-00-00-00-8C-B9-00-00
A	10:32:52.149	"(0x) 00-00-00-00-00-00-00-00-8C-B9-00-00" sent
I	10:32:52.178	Notification received from 00001531-1212-efde-1523-785feabcd123, value: (0x) 10-01-03
A	10:32:52.178	"Response for: Start DFU
Value: Not supported" received
V	10:32:52.179	Writing request to characteristic 00001531-1212-efde-1523-785feabcd123
D	10:32:52.179	gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123, value=0x01)
I	10:32:52.207	Data written to 00001531-1212-efde-1523-785feabcd123, value: (0x) 01
A	10:32:52.207	"Start DFU" sent
V	10:32:52.207	Writing command to characteristic 00001532-1212-efde-1523-785feabcd123
D	10:32:52.207	gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123, value=0x8CB90000)
I	10:32:52.209	Data written to 00001532-1212-efde-1523-785feabcd123, value: (0x) 8C-B9-00-00
A	10:32:52.209	"(0x) 8C-B9-00-00" sent

philips77, 2017-03-22:
And how about adding some delays here and there? For example before every write. I assume it won't work either. Did you check with firmware developers about my long comment above?

andy-fueled, 2017-03-22:
I added delays of 1 second and 3 second, but still it did not work. 
Mike is going to talk with the firmware developers today, and he will tell them everything from your messages, and try to get the bootloader hex file.

philips77, 2017-04-03:
Hi Andy, any news?

andy-fueled, 2017-04-03:
@philips77 It seems like this is not the top priority for the firmware people, so this issue is on hold in the meantime

mostafaberg, 2017-08-08:
@andy-fueled I'll close this issue since it's been stale since April :) if you require more help please comment here and we'll reopen and take a look again

closed 2017-08-08 by mostafaberg