nordicsemi/IOS-DFU-Library #97: DFU update fails with no error

jotaefejota, 2017-03-16:
Hi @mostafaberg , I have a very similar issue to #85 (possibly the exact same one).

See my logs:

> 
> DFU Log 5: Connected to device
> DFU Log 1: Discovering services...
> DFU Log 0: peripheral.discoverServices(nil)
> DFU Log 5: Services discovered
> DFU Log 1: Starting Legacy DFU...
> DFU Log 5: Connected to device
> DFU Log 5: Services discovered
> DFU Log 1: Legacy DFU Service found
> DFU Log 1: Discovering characteristics in DFU Service...
> DFU Log 0: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
> DFU Log 5: DFU characteristics discovered
> DFU Log 1: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
> DFU Log 1: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: DFU Control Point notifications enabled
> DFU Log 15: Application with buttonless update found
> DFU Log 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.writeValue(0xd7ae, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
> DFU Log 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

(and nothing else happens)

iOSDFULibrary v3.0.6
Bootloader S130 1.0.0
Softdevice S130 1.0.0
Nrf51 S130 1.0.0

Any pointers?
Thanks

mostafaberg, 2017-03-16:
Thanks!, What happens after the last log ? does the peripheral turn off and restart on bootloader mode or does it stay as is ?

jotaefejota, 2017-03-16:
I'm working on Xcode and I don't have access to debugging tools for the chipset (or the knowledge) if that's what you are referring to. 
Otherwise, no logging on Xcode.

mostafaberg, 2017-03-16:
you don't need specialized debugging tools, can you see if the peripheral begins broadcasting a different name after sending that command ? or at least if it disconnects from the phone ?

jotaefejota, 2017-03-17:
There is no more logging on Xcode after that. The CBCentralManager is stopped and the chipset seems to stay on a "waiting state" forever. I don't see a new name broadcasted and I don't think is disconnected from the phone but I can't confirm it (any suggestions on how to know whether is disconnected?). 

I've done an experiment though:

On DFUServiceInitiator.swift > public init(centralManager: CBCentralManager, target: CBPeripheral) > I have commented self.centralManager.stopScan() to see if I could get more info from the logs (I realise this is not a solution). And on my app, I've also let the centralManager run after pairing with the DEVICE.

The bootloader is then found and it appears to start the process, but it gets to a point where it doesn't progress and no more logs are shown:

> Fimrware update started
> DFU Log 5: Connected to DEVICE
> DFU Log 1: Discovering services...
> DFU Log 0: peripheral.discoverServices(nil)
> DFU Log 5: Services discovered
> DFU Log 1: Starting Legacy DFU...
> DFU Log 5: Connected to DEVICE
> DFU Log 5: Services discovered
> DFU Log 1: Legacy DFU Service found
> DFU Log 1: Discovering characteristics in DFU Service...
> DFU Log 0: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
> DFU Log 5: DFU characteristics discovered
> DFU Log 1: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
> DFU Log 1: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: DFU Control Point notifications enabled
> DFU Log 15: Application with buttonless update found
> DFU Log 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
> DFU Log 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
> DFU Log 5: DFU Bootloader found with name DEVICE
> DFU Log 1: Connecting to DEVICE...
> DFU Log 0: centralManager.connect(peripheral, options: nil)
> DFU Log 0: [Callback] Central Manager did connect peripheral
> DFU Log 5: Connected to DEVICE
> DFU Log 1: Discovering services...
> DFU Log 0: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
> DFU Log 5: Services discovered
> DFU Log 1: Legacy DFU Service found
> DFU Log 1: Discovering characteristics in DFU Service...
> DFU Log 0: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
> DFU Log 5: DFU characteristics discovered
> DFU Log 1: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
> DFU Log 1: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: DFU Control Point notifications enabled
> DFU Log 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
> DFU Log 1: Writing image sizes (0b, 0b, 56168b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
> DFU Log 0: peripheral.writeValue(0x000000000000000068db0000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
> DFU Log 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
> DFU Log 10: Start DFU (Op Code = 1, Upload Mode = 4) request sent

mostafaberg, 2017-03-17:
Does this also reproduce with the nRF Toolbox app ? can you give it a try and see if the same issue occurs ?

jotaefejota, 2017-03-17:
I believe so:

When trying to upload the new .hex firmware on application mode:

[image]

And the logs from the nRF Toolbox app are:

> file:///private/var/mobile/Containers/Data/Application/39786B26-D71D-41DA-9970-97F0BD207C2F/Documents/DEVICE_0_99_7_connect.hex
> Verbose: Connecting to DEVICE...
> Debug: centralManager.connect(peripheral, options: nil)
> Debug: [Callback] Central Manager did connect peripheral
> Info: Connected to DEVICE
> Verbose: Discovering services...
> Debug: peripheral.discoverServices(nil)
> Info: Services discovered
> Verbose: Starting Legacy DFU...
> Info: Connected to DEVICE
> Info: Services discovered
> Verbose: Legacy DFU Service found
> Verbose: Discovering characteristics in DFU Service...
> Debug: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
> Info: DFU characteristics discovered
> Verbose: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
> Debug: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
> Verbose: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
> Application: DFU Control Point notifications enabled
> Warning: Application with buttonless update found
> Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> Debug: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
> Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
> Application: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
> 

And again, the app UI stays on "Enabling DFU Bootloader..." and no more messages logged on nRF Toolbox project on Xcode.

philips77, 2017-03-17:
Hi again. In nrf toolbox you may force dfu in settings (normal ios settings -> nrf toolbox -> external MCU DFU -> on). I know... Not very obvious and user friendly, but it's a rare case.

jotaefejota, 2017-03-17:
Did that, run the nRF Toolbox project from Xcode again and got this:

> file:///private/var/mobile/Containers/Data/Application/0EE56D9E-05E2-4979-903D-7A846A7F5CD5/Documents/DEVICE_0_99_7_connect.hex
> Verbose: Connecting to DEVICE...
> Debug: centralManager.connect(peripheral, options: nil)
> Debug: [Callback] Central Manager did connect peripheral
> Info: Connected to DEVICE
> Verbose: Discovering services...
> Debug: peripheral.discoverServices(nil)
> Info: Services discovered
> Verbose: Starting Legacy DFU...
> Info: Connected to DEVICE
> Info: Services discovered
> Verbose: Legacy DFU Service found
> Verbose: Discovering characteristics in DFU Service...
> Debug: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
> Info: DFU characteristics discovered
> Verbose: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
> Debug: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
> Verbose: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
> Application: DFU Control Point notifications enabled
> Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> Debug: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
> Verbose: Writing image sizes (0b, 0b, 56092b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
> Debug: peripheral.writeValue(0x00000000000000001cdb0000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
> Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
> Application: Start DFU (Op Code = 1, Upload Mode = 4) request sent

The app sits reading: "Starting DFU" and no more logs are shown on Xcode.

philips77, 2017-03-17:
Ok, some questions:
1. Which SDK is your project forked from?
2. Do you support buttonless jump or should the DFU be done immediately in the first connection?
3. Do you have DFU Version characteristic in the app and bootloader modes and what values they return?

philips77, 2017-03-17:
SD S130 1.0.0 was used in SDK 10, as far as I remember. Then you should have the DFU Version char in both app and bootloader mode. In app mode it should be returning 0x0100 to let the library know that buttonless jump is supported, while in bootloader mode the value returned should be something like 0x0800, which means that fw signing is supported.
However, in your case I see that the DFU Version char is not found, so the lib behaves as if you were using SDK 6.1 (then there was no such characteristic) and it bases the action (to jump or not to jump, this is the question) on number of services (only DFU = bootloader mode, or more services = app mode and forceDfu parapeter is used to determine should the app jump (forceDfu=false), or start DFU (forceDfu=true).

jotaefejota, 2017-03-17:
It turns out the process was stopped because of our own flow implementation of the DFU. 
You've been really helpful on focussing me towards the problem. 
Feel free to close this issue. 
Many thanks!

philips77, 2017-03-17:
Glad to hear that!

closed 2017-03-17 by philips77