nordicsemi/IOS-DFU-Library #100: Unable to reconnect to ble device after uploading new firmware.

cjlarsen, 2017-04-04:
Complete updating firmware to a nrf51422, DFU library disconnects.  I attempt to scan and then discover the same device, but am unable to reconnect.  Device is never found.

philips77, 2017-04-04:
1. Did you remember to restore the delegates to CBPeripheral and CBCentralManager?
2. Are you sure your device is advertising after the update? Do you see it in nrf connect? Is the app compatible with the SoftDevice?

cjlarsen, 2017-04-04:
@philips77 We are sure it is SoftDevice compatible. I think it's the delegates, attempting this fix in swift: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/29#issuecomment-252816614

cjlarsen, 2017-04-04:
@philips77 Yep it was the delegate being stolen, closing this.  Thanks.

closed 2017-04-04 by cjlarsen