nordicsemi/IOS-DFU-Library #101: Objective C Error DFU  error 202

AnD010, 2017-04-06:
Hi, I try to update the firmware to my device, but don´t work fine... sometimes work fine others no. 
Generally works well when I try the process some times .
I followed the steps to use in objective-c the dfu library
```
Using Zip (0.7.0)
Using iOSDFULibrary (3.0.6)
```
The steps I perform are as follows:
- create a queue 
`dispatch_queue_t centralQueue = dispatch_queue_create ("com.BLE.UPDATEDFU", DISPATCH_QUEUE_CONCURRENT);`

- when centralManagerDidUpdateState is called
  `[Self.centralManager scanForPeripheralsWithServices: @ [[CBUUID UUIDWithString: LEGACY_DFU_SERVICE]] options: @ {CBCentralManagerScanOptionAllowDuplicatesKey: @NO}];`
-When didDiscoverPeripheral is called 
`[Self.centralManager connectPeripheral: self.dfuPeripheral options: nil];`
-When didConnectPeripheral is called
```
  Self.selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile: url];
    [Self startDFUProcess];
```
 Where I do
```
DFUServiceInitiator * initiator = [[DFUServiceInitiator alloc] initWithCentralManager: self.centralManager target: self.dfuPeripheral];
    [Initiator setLogger: self];
    [Initiator setDelegate: self];
    [Initiator setPacketReceiptNotificationParameter: 12];
    Initiator.forceDfu = NO;
    [Initiator setProgressDelegate: self];
    
    // This enables the experimental Buttonless DFU feature from SDK 12.
    // Please, read the field documentation before use.
    Initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
    
    Self.dfuController = [initiator withFirmware: self.selectedFirmware] .start ;
```

When I get wrong  result 
```
2017-04-06 19:38:51.546 DriveSmartLibraryIOS_BLE[4959:2474184] advertisementData <CBPeripheral: 0x125d25480, identifier = AF06B94C-0689-6A0C-0054-EFECA611F254, name = DfuTest, state = disconnected> at -57 advertisementData {
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataLocalName = DfuTest;
    kCBAdvDataServiceUUIDs =     (
        "00001530-1212-EFDE-1523-785FEABCD123"
    );
}
2017-04-06 19:38:51.547 LibraryIOS_BLE[4959:2474184] DFUUUUUUUUUUUUUUUUUUUU
2017-04-06 19:38:51.686 LibraryIOS_BLE[4959:2474174] Connected
2017-04-06 19:38:51.745 LibraryIOS_BLE[4959:2474131] DFUDELEGATE DFUServiceDelegate dfuStateDidChangeT 0
2017-04-06 19:38:51.746 LibraryIOS_BLE[4959:2474131] Successfully received the UpdateDFUNotification notification!
2017-04-06 19:38:51.747 LibraryIOS_BLE[4959:2474174] DFUDELEGATE LoggerDelegate logWithLevel Connected to DfuTest
2017-04-06 19:38:51.747 LibraryIOS_BLE[4959:2474174] DFUDELEGATE LoggerDelegate logWithLevel Discovering services...
2017-04-06 19:38:51.747LibraryIOS_BLE[4959:2474174] DFUDELEGATE LoggerDelegate logWithLevel peripheral.discoverServices(nil)
2017-04-06 19:38:52.009 LibraryIOS_BLE[4959:2474166] DFUDELEGATE LoggerDelegate logWithLevel [Callback] Central Manager did disconnect peripheral with error: The connection has failed unexpectedly.
2017-04-06 19:38:52.012 LibraryIOS_BLE[4959:2474166] DFUDELEGATE LoggerDelegate logWithLevel Disconnected
2017-04-06 19:38:52.012 LibraryIOS_BLE[4959:2474166] DFUDELEGATE LoggerDelegate logWithLevel Error 10: The connection has failed unexpectedly.
2017-04-06 19:38:52.013 LibraryIOS_BLE[4959:2474131] DFUDELEGATE DFUServiceDelegate didOccurWithMessage 202
2017-04-06 19:38:52.013 LibraryIOS_BLE[4959:2474131] DFUDELEGATE DFUServiceDelegate didOccurWithMessage The connection has failed unexpectedly. (code: 10)
```

When I'm doing well I use DuplicatesKey: @YES, but not work fine always....
` [Self.centralManager scanForPeripheralsWithServices: @ [[CBUUID UUIDWithString: LEGACY_DFU_SERVICE]] options: @ {CBCentralManagerScanOptionAllowDuplicatesKey: @YES}];`

```
2017-04-06 19:30:59.130 LibraryIOS_BLE[4930:2471056] advertisementData <CBPeripheral: 0x127e845e0, identifier = AF06B94C-0689-6A0C-0054-EFECA611F254, name = DfuTest, state = disconnected> at -57 advertisementData {
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataLocalName = DfuTest;
    kCBAdvDataServiceUUIDs =     (
        "00001530-1212-EFDE-1523-785FEABCD123"
    );
}
[... 185 lines cut ...]
2017-04-06 19:31:12.139 LibraryIOS_BLE[4930:2471050] DFUDELEGATE LoggerDelegate logWithLevel Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100301
2017-04-06 19:31:12.139 LibraryIOS_BLE[4930:2471050] DFUDELEGATE LoggerDelegate logWithLevel Response (Op Code = 3, Status = 1) received
2017-04-06 19:31:12.139 LibraryIOS_BLE[4930:2471050] DFUDELEGATE LoggerDelegate logWithLevel Upload completed in 9.71 seconds
2017-04-06 19:31:12.140 LibraryIOS_BLE[4930:2470856] DFUDELEGATE DFUServiceDelegate dfuStateDidChangeT 4
2017-04-06 19:31:12.140 LibraryIOS_BLE[4930:2470856] Successfully received the UpdateDFUNotification notification!
2017-04-06 19:31:12.141 LibraryIOS_BLE[4930:2471050] DFUDELEGATE LoggerDelegate logWithLevel Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-04-06 19:31:12.142 LibraryIOS_BLE[4930:2471050] DFUDELEGATE LoggerDelegate logWithLevel peripheral.writeValue(0x04, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-04-06 19:31:12.243 LibraryIOS_BLE[4930:2471057] DFUDELEGATE LoggerDelegate logWithLevel Data written to 00001531-1212-EFDE-1523-785FEABCD123
2017-04-06 19:31:12.244 LibraryIOS_BLE[4930:2471057] DFUDELEGATE LoggerDelegate logWithLevel Validate Firmware (Op Code = 4) request sent
2017-04-06 19:31:12.248 LibraryIOS_BLE[4930:2471057] DFUDELEGATE LoggerDelegate logWithLevel Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100401
2017-04-06 19:31:12.249 LibraryIOS_BLE[4930:2471057] DFUDELEGATE LoggerDelegate logWithLevel Response (Op Code = 4, Status = 1) received
2017-04-06 19:31:12.250 LibraryIOS_BLE[4930:2470856] DFUDELEGATE DFUServiceDelegate dfuStateDidChangeT 5
2017-04-06 19:31:12.251 LibraryIOS_BLE[4930:2470856] Successfully received the UpdateDFUNotification notification!
2017-04-06 19:31:12.253 LibraryIOS_BLE[4930:2471057] DFUDELEGATE LoggerDelegate logWithLevel Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-04-06 19:31:12.269 LibraryIOS_BLE[4930:2471057] DFUDELEGATE LoggerDelegate logWithLevel peripheral.writeValue(0x05, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-04-06 19:31:12.319 LibraryIOS_BLE[4930:2471120] DFUDELEGATE LoggerDelegate logWithLevel Data written to 00001531-1212-EFDE-1523-785FEABCD123
2017-04-06 19:31:12.324 LibraryIOS_BLE[4930:2471120] DFUDELEGATE LoggerDelegate logWithLevel Activate and Reset (Op Code = 5) request sent
2017-04-06 19:31:12.484 LibraryIOS_BLE[4930:2470856] *** -[NSKeyedUnarchiver initForReadingWithData:]: data is NULL
2017-04-06 19:31:12.539 LibraryIOS_BLE[4930:2471115] DFUDELEGATE LoggerDelegate logWithLevel [Callback] Central Manager did disconnect peripheral
2017-04-06 19:31:12.540 LibraryIOS_BLE[4930:2471115] DFUDELEGATE LoggerDelegate logWithLevel Disconnected by the remote device
2017-04-06 19:31:12.540 LibraryIOS_BLE[4930:2470856] DFUDELEGATE DFUServiceDelegate dfuStateDidChangeT 6
2017-04-06 19:31:12.541 LibraryIOS_BLE[4930:2470856] Successfully received the UpdateDFUNotification notification!
```

Thanks!

AnD010, 2017-04-06:
A note .. my device is iphone 5s 9.2 version
other note.. with app in Android nRF Connect always works fine....

rushstrike, 2017-04-08:
I dont know if this will help. But I had trouble with DFU sometimes working and sometimes not. My solution was to make sure that you disconnect the connected peripheral first then add a 3-5 sec to make sure it's disconnected before starting DFU.

`[self.centralManager cancelPeripheralConnection:self.connectedPeripheral];`
wait 5 sec before you start DFU

Let me know if this helps.

AnD010, 2017-04-09:
Hi @rushstrike , thanks for your advice, but don´t work for me...
Now I have a implementation that work fine.... the steps to follow are these:
I have a main queue that is the one that worked obtaining data of the advertising, now when detected  that enters my element in dfu  in 
`- (void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {`

I call another  queue process:
`  dispatch_queue_t centralQueue = dispatch_queue_create("com.BLE.UPDATEDFU", DISPATCH_QUEUE_CONCURRENT);`
When this process detects dfu in 
`- (void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {`

I start the upgrade process, I never connect the peripheral or anything, only when I detect it in the above method I start the upgrade process and it works for me.
The upgrade code is like the one above but I do not make the connection with the peripheral.

Regards

closed 2017-04-09 by AnD010