nordicsemi/IOS-DFU-Library #102: best practice before running DFU

rushstrike, 2017-04-08:
Is it expected that we disconnect from the peripheral first before we start DFU using:

[self.centralManager cancelPeripheralConnection:self.connectedPeripheral];

I noticed if I don't disconnect then the DFU update only works sometimes.

mostafaberg, 2017-04-10:
No you shouldn't need to disconnect, but even if you do that should affect anything. the library takes that into consideration of the peripheral is connected or not.

When you say that it works only sometimes, can you clarify a bit more on what goes wrong ?
any logs of the failing process would help us out a lot.

rushstrike, 2017-04-10:
When I leave the device connected calling the dfu start process does not result in the device restarting to the boot loader. However it does automatically disconnect from the peripheral. Then with my retry logic it would succeed in the 2nd retry. If I start dfu update process with the peripheral already disconnected then it would work on the first try.

mostafaberg, 2017-04-10:
We will need to see the log from the attempt that fails and the attempt that succeeds so we can help you further

mostafaberg, 2017-04-12:
Any updates on that issue ? let us know if we can be of any help, if we don't hear back within a couple of days we will assume you resolved the issue 👍

mostafaberg, 2017-04-20:
I will close this issue as we haven't heard feedback over a week so will assume it's fixed. if not, please reopen.

closed 2017-04-20 by mostafaberg