nordicsemi/IOS-DFU-Library #106: withFirmware: warn_unused_result warning in XCode 8.3.1 (8E1000a)

casey-pax, 2017-04-13:
The [DFUServiceInitiator withFirmware:] is marked with a SWIFT_WARN_UNUSED_RESULT. However, the comments claim this is for chaining. Nor is it clear why you would have to use the return result.

But the latest XCode now has a build warning. Please remove the SWIFT_WARN_UNUSED_RESULT.

mostafaberg, 2017-04-20:
Can you show the whole code snippet ?, when we mean chaining we mean calling start directly after creating the initiator on the same line.

casey-pax, 2017-04-20:
Sure, but if it's for code chaining, it should declare SWIFT_WARN_UNUSED_RESULT. The caller should have the option not to use the returned result.

[image]

[image]

mostafaberg, 2017-04-21:
What you are looking at is generated code and we did not write that flag ourselves, however, it's a good idea to follow warnings and treat them as errors, what I would say is if you don't want to chain you may just set the result again to `self.dfuInitiator` and you should be fine.

This is not ideal, I will take a look if we can have an improvement to remove that chaining feature or at least support only one way to instantiate, there's no need for all that hassle I agree.

let me know if setting the result like that fixes it for you

```
self.dfuInitiator = [_dfuInitiator withFirmware:selectedFirmware];
```

casey-pax, 2017-04-21:
I've confirmed this will resolve the issue:
@discardableResult public func with(firmware file: DFUFirmware) -> DFUServiceInitiator {

I can make a PR if you like.

mostafaberg, 2018-06-19:
This issue is too stale, will close for now until further need appears

closed 2018-06-19 by mostafaberg