nordicsemi/IOS-DFU-Library #107: DFU gets stuck when applications is put to the background

bisikli, 2017-04-14:
Hello all, 

I have integrated iOSDFULibrary into my project and can upload the my firmware to the device just fine if i dont do anything during uploading process. However during monkey testing i realised, if i put the app into background and open again during uploading stage, DFU gets stuck and stays at uploading state forever - gives no timeout error or whatsoever. If i try to abort the controller, dfu state do not changed to Aborted (i am checking dfuStateDidChangeTo callback for this.)

mostafaberg, 2017-04-20:
Hi @bisikli !, sorry for the late response we had easter vacations :)

I cannot seem to reproduce the issue. firmware does not continue uploading in the background, but as soon as the app comes tot he foreground it does resume as expected.

How are you managing the BLE connection ? you might be having some issue there.

Perhaps it also has something to do with enabling background mode ?
In Xcode go to the app capabilities, and check if *Background Modes* -> *Uses Bluetooth LE accessories* is on or off, try toggling that and see if your issue is resolved.

bisikli, 2017-04-20:
Oh OK, that solves the problem thanks!

closed 2017-04-20 by bisikli