nordicsemi/IOS-DFU-Library #108: Is this compatible with SDk v13.0.0?

pruthvikar, 2017-04-20:
I was wondering if the library needs to be changed to work with SDK v13.0.0.

I am aware that the buttonless DFU service and characteristic have changed UUIDs, and that there are extended error codes. I was wondering if that means this library at v3.1.0 is still compatible?

mostafaberg, 2017-04-20:
I'm not sure i'm following the question, do  you mean to ask if v3.1.0 is compatible with SDK v13 ? or if earlier versions are compatitble with SDK 13 ?

pruthvikar, 2017-04-20:
Is version v3.1.0 compatible with SDK v13?

mostafaberg, 2017-04-20:
Yes it is, you can see that in the release here :)
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases/tag/3.1.0

pruthvikar, 2017-04-20:
Oh, perfect, thanks!

closed 2017-04-20 by pruthvikar

mostafaberg, 2017-04-20:
No worries :) lest us know if you have any more questions!