nordicsemi/IOS-DFU-Library #109: Update stuck on "Connecting" for DFU with bootloader + application

anshulmittal712, 2017-04-26:
Refer attached video.

When I first try to update, it throws an error saying "Device disconnected unexpectedly".
I then hit the update button again. This time it starts on to upload the bootloader, finishes that (30-second mark on the attached video), and then instead of starting with the application upload, just remains stuck at "Connecting".

The .zip is fine as the same works as expected from an Android device.

Video: [Link to Youtube](https://youtu.be/Uq74WW5-CUg)

philips77, 2017-04-26:
Hi, thanks for the report. I have some questions:

1. Could you post here the log? You'll get it by tapping the Share button on tht dfu screen.
2. Which sdk is your old and new bootloader from?
3. Are you sure the new bootloader advertises with the same Service uuid? For example, if the old bootloader was a legacy one (sdk 11 or older), the new one also has to be a legacy one and the adv packet must contain the legacy dfu service uuid. If the new bootloader is from sdk 12 or newer (secure dfu) you must first update only the bootloader and then start dfu again with app update only. This is because the ios lib assumes both bootloader have the same type.
4.Could you try with nRF Toolbox app on ios?

mostafaberg, 2017-04-26:
Ok this is bad, i would say there is an error with the operation but the app is not displaying it, please try what @philips77 says and let us know if you have the same issue with the nRF Toolbox app so we can rule out if it's the app that has an issue or if it's the DFU Library. thanks

philips77, 2017-04-26:
Remember, that on iOS the device address is not available, so the lib expects the dfu service uuid in the adv packet instead of relying on the address, as on Android. You may change the default filter when you use the lib in your own project, but not in nrf connect or nrf toolbox.

mostafaberg, 2017-04-27:
@jovirus

jovirus, 2017-04-27:
Hi, 
When you update the bootloader to a device, the address of the new bootloader might have changed, but the DFU lib tries to reconnect the older device, therefor it will failed. The app is designed not to handle such situation, it requires a update from the DFU lib.

philips77, 2017-04-27:
Hi @anshulmittal712 ,
I know the reason why the DFU stucked at Connecting on iOS.
When you update only the bootloader (without SoftDevice), the old application is not being removed, as there is enough space for double bank update of the bootloader. So when the first upload is complete and the new bootloader was flashed successfully, the device with restart and... the old app will be started. It will start advertise, but as the library expects by default the DFU Service UUID in the adv packet, it won't connect, as such device won't be found.

That is one of the reasons the DfuServiceInitiator allows to specify the device selector: [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L71), so the final app that uses the DFU lib should set the selector that would select a correct device (based on the app's adv packet.
The nRF Connect, or nRF Toolbox, use the default selector which expects the DFU service UUID and it won't work when you do a single update with bootloader+application. If you want to flash your app using one of those apps, you have to split the ZIP into 2: one with BL update only, and the other one with App update only and perform both updates one after another manually.

mostafaberg, 2017-04-28:
@philips77  I think this is an issue our library should handle. a Bootloader+Application update should flash bootloader,on success the device restarts application, then we should put it back in bootloader mode and flash the rest of the firmware, doesn't that make more sense ?

does that make sense ? if so we can get some fix out, if i'm missing something, let me know :)

philips77, 2017-04-28:
Well... There is a chance that a fix could be made, but it also won't work 100%. Let's say the was an app before the update. If the buttonless update is used at first I can keep the original CBPeripheral and if i know (based on the info from manifest file) that I sent the bootloader, I can connect to the original one, instead of scanning. But if user switched to bl mode with the button at the first place, and after flashing new bl the old app will start, we have no clue how to connect to it.
If say that the Selector was made for this purpose... The similar problem you get when you update legacy dfu bootloader to secure one (or the other way round)
 (assuming they have been made to have the same starting address in nrf memory). Selection of dfu type is done once in iOS, so updating sd+bl will work, but following app update won't work, do 2 zip files are required.

mostafaberg, 2017-04-28:
I didn't have the hardware DFU option in mind, I see what you're saying, absolutely right!
We should close this for now then I guess as the only solution is to split it as you said.

If we have more updates on that we will update this thread :)

@anshulmittal712 Your only fix for now, unfortunately is to split the firmware parts into 2 Zip files until we come up with a better solution.

if you have an issue generating those files please open up a new issue and we will help you out there.

sorry for the inconvenience

closed 2017-04-28 by mostafaberg

anshulmittal712, 2017-04-28:
Hi, sorry I was away. I'll try to publish the logs here soon.
And yes, if it helps anyone else - that's the current workaround I'm doing. I've split it into 2 .zip files, one only with the bootloader, and other just with the application, and am doing them sequentially. (calling the second on the .completed of first)

mostafaberg, 2017-04-28:
Perfect !, we will do our best to find a simpler way. keep us updated, logs would be very valuable!
have a great day