nordicsemi/IOS-DFU-Library #110: Error 202, Device disconnected unexpectedly

Shanti-SR, 2017-05-03:
Hello,
I can not progress from the DFUStateEnablingDfuMode state. I've tested the app "nRF Toolbox" and "nRF Connect" but I always get the error that the device has been disconnected. 

With the [IOS-Pods-DFU-Library](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library) project, I also have the same error.

This is my log:
2017-05-03 14:07:22:429 Dev[2856:840190] Connected to nRF5x
2017-05-03 14:07:22:430 Dev[2856:840190] Services discovered
2017-05-03 14:07:22:430 Dev[2856:840190] Starting Legacy DFU...
2017-05-03 14:07:22:430 Dev[2856:840190] Connected to nRF5x
2017-05-03 14:07:22:430 Dev[2856:840190] Services discovered
2017-05-03 14:07:22:430 Dev[2856:840190] Legacy DFU Service found
2017-05-03 14:07:22:430 Dev[2856:840190] Discovering characteristics in DFU Service...
2017-05-03 14:07:22:430 Dev[2856:840190] peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
2017-05-03 14:07:22:567 Dev[2856:840190] DFU characteristics discovered
2017-05-03 14:07:22:570 Dev[2856:840190] Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
2017-05-03 14:07:22:570 Dev[2856:840190] peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
2017-05-03 14:07:22:686 Dev[2856:840190] Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
2017-05-03 14:07:22:687 Dev[2856:840190] DFU Control Point notifications enabled
2017-05-03 14:07:22:687 Dev[2856:840190] Application with buttonless update found
2017-05-03 14:07:22:687 Dev[2856:840190] Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-05-03 14:07:22:688 Dev[2856:840190] peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-05-03 14:07:23:460 Dev[2856:840190] [Callback] Central Manager did disconnect peripheral
2017-05-03 14:07:23:460 Dev[2856:840190] Disconnected by the remote device
2017-05-03 14:07:23:460 Dev[2856:840190] Connecting to nRF5x...
2017-05-03 14:07:23:460 Dev[2856:840190] centralManager.connect(peripheral, options: nil)
2017-05-03 14:09:25:374 Dev[2856:840190] [Callback] Central Manager did connect peripheral
2017-05-03 14:09:25:374 Dev[2856:840190] Connected to nRF5x
2017-05-03 14:09:25:374 Dev[2856:840190] Discovering services...
2017-05-03 14:09:25:375 Dev[2856:840190] peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
2017-05-03 14:09:26:009 Dev[2856:840190] Services discovered
2017-05-03 14:09:26:010 Dev[2856:840190] Legacy DFU Service found
2017-05-03 14:09:26:010 Dev[2856:840190] Discovering characteristics in DFU Service...
2017-05-03 14:09:26:010 Dev[2856:840190] peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
2017-05-03 14:09:26:131 Dev[2856:840190] DFU characteristics discovered
2017-05-03 14:09:26:131 Dev[2856:840190] Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
2017-05-03 14:09:26:131 Dev[2856:840190] peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
2017-05-03 14:09:26:250 Dev[2856:840190] Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
2017-05-03 14:09:26:250 Dev[2856:840190] DFU Control Point notifications enabled
2017-05-03 14:09:26:250 Dev[2856:840190] wait(1000)
2017-05-03 14:09:27:331 Dev[2856:840190] Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-05-03 14:09:27:331 Dev[2856:840190] peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-05-03 14:09:27:333 Dev[2856:840190] Writing image sizes (0b, 0b, 50000b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
2017-05-03 14:09:27:333 Dev[2856:840190] peripheral.writeValue(0x000000000000000050c30000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
2017-05-03 14:09:28:103 Dev[2856:840190] [Callback] Central Manager did disconnect peripheral
2017-05-03 14:09:28:103 Dev[2856:840190] Disconnected by the remote device
2017-05-03 14:09:28:104 Dev[2856:840190] DFUError 202, message: Device disconnected unexpectedly

Thanks for any help.

mostafaberg, 2017-05-03:
Hi !, how are you creating the firmware ? is it possible to send us a copy hex that reproduces the issue ?

Shanti-SR, 2017-05-03:
Hi. I honestly do not know how the firmware is created, another programmer created the firmware. This is the firmware with which I have problems:
https://www.dropbox.com/s/00az4glkni6dybi/Ex1.5.3DFU.zip?dl=0

mostafaberg, 2017-05-05:
Let me check if i can reproduce that issue with your firmware on Monday, which board are you testing on ?

Shanti-SR, 2017-05-06:
In the end I have been able to solve the problem. When the peripheral is disconnected, I had to reconnect to a peripheral called "DfuTarg". Thanks for the help also.

closed 2017-05-06 by Shanti-SR