nordicsemi/IOS-DFU-Library #111: Issue in getting serial number of BLE device

lovekeshsaxena, 2017-05-06:
I am working on a project in which i have to get the readings from device (BLE device). I have two devices BP and BG. I can successfully get the data from the devices using your library. I also want the serial number and the MAC address of both the devices to be fetched but unable to get them. Please help me to get the device serial number and the MAC address.

philips77, 2017-05-06:
Hi, iOS api does not allow to get the address. You'd have to implement a characteristic that returns it if you need to have it on iOS.

lovekeshsaxena, 2017-05-08:
I was trying to get serial number of device by it's information service with serial number characteristic. I am able to get information service, but it does not contain serial number characteristic.
In android, able to get serial number for the device.
Is there any way to get serial  number from device when service does not contain serial number characteristic? 
It's basic requirement for app,  to verify Bluetooth device.

mostafaberg, 2017-05-08:
This is not an issue, and it's not related to DFU library, please post your questions at our [DevZone](https://devzone.nordicsemi.com) instead :)

closed 2017-05-08 by mostafaberg