nordicsemi/IOS-DFU-Library #113: Incorrect Command sequence

pruthvikar, 2017-05-18:
According to the [documentation here](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v13.0.0%2Flib_dfu_transport_ble.html&cp=4_0_0_3_4_2_2_3_0&anchor=lib_dfu_transport_msc_init), The sequence of commands sent by the Controller must begin with a `Select command` followed by a `Create command: size`.

The implementation of this library however jumps straight into a `Create command: size` causing the DFU process to fail, as shown in the device log here

```
:DEBUG:Waiting for events
DFU_REQ_HANDLING:DEBUG:In nrf_dfu_command_req()
DFU_REQ_HANDLING:DEBUG:req_type = WRITE
DFU_REQ_HANDLING:DEBUG:req_len = 2
DFU_REQ_HANDLING:DEBUG:p_req[0] = 0x01
DFU_REQ_HANDLING:DEBUG:p_req[1] = 0x00
DFU_REQ_HANDLING:INFO:Before OP write command
DFU_REQ_HANDLING:ERROR:Error. Init command larger than expected. 
DFU_BLE:INFO:Failure to run packet write
```

The Android DFU library does not have the same issue as it seems to correctly follow the implementation guidelines. Here is a similar log using the Android library (truncated at end)

```
:DEBUG:Waiting for events
DFU_BLE:INFO:Received select object
DFU_REQ_HANDLING:DEBUG:In nrf_dfu_command_req()
DFU_REQ_HANDLING:DEBUG:req_type = SELECT
DFU_REQ_HANDLING:INFO:Valid Command: NRF_DFU_OBJECT_OP_SELECT
DFU_BLE:INFO:Sending Object Info: [0x60, 0x06, 0x01 max: 0:x00000100 0:x00000000, CRC:0x00000000]
DFU_BLE:INFO:Set receipt notif
DFU_BLE:INFO:Sending Response: [0x2, 0x1]
DFU_BLE:INFO:Received create object
DFU_REQ_HANDLING:DEBUG:In nrf_dfu_command_req()
DFU_REQ_HANDLING:DEBUG:req_type = CREATE
DFU_REQ_HANDLING:DEBUG:obj_type = COMMAND
```

mostafaberg, 2017-07-21:
I took a look at that now, the library does actually send the select command (It's called read object info command in the library) and it' the first actoin that gets sent.

which version of the library are you using ? and is the peripheral completely reset at that point ? it might be just resuming an earlier stage

can you give us the exact version of iOSDFULibrary you're using and what's your setup (Firmware, SD, BL versions) on the peripheral side ?

mostafaberg, 2017-08-08:
@pruthvikar Do you still require help regarding this issue ? if not i'll close the issue during the week :) if you see this too late just comment and i'll reopen the issue

closed 2017-08-18 by mostafaberg