nordicsemi/IOS-DFU-Library #114: xcode8.3.1   oc project

JackLSH2016, 2017-05-25:
hello,I use oc in xcode8.3.1,and I can't find Use Legacy Swift version this select.How can I do something to use IOS-Pods-DFU-Library?

BabaYB, 2017-05-26:
I didn't find in Xcode 8.3.2 too, but there is no error when I build my project.How about you?
And my app broken at running controller = [initiator start];

JackLSH2016, 2017-05-26:
it's ok now,maybe you need to save the controller as a property.

BabaYB, 2017-05-26:
Yeah,I did do that.So dfu in your app works now?This is my test code, have no ideal, I don't know what the problem is.THX 

    DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager: _centralMgr target:_discoveredPeripheral];
    initiator = [initiator withFirmware:_selectedFirmware];
    initiator.forceDfu = [[[NSUserDefaults standardUserDefaults] valueForKey:@"dfu_force_dfu"] boolValue];
    initiator.packetReceiptNotificationParameter = [[[NSUserDefaults standardUserDefaults] valueForKey:@"dfu_number_of_packets"] intValue];
    initiator.logger = self;
    initiator.delegate = self;
    initiator.progressDelegate = self;
    initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
    self.dfuController = [initiator start];

mostafaberg, 2017-05-29:
Please follow the exact steps here to integrate the library with Obj-C:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/README_OBJC.md

If you have further questions regarding using the library please post the questions to our devzone so you'll get quicker answers and better support

closed 2017-05-29 by mostafaberg