nordicsemi/IOS-DFU-Library #117: DFU updates wrong nrf52 device

mcclen, 2017-06-07:
I'm a novice when it comes to Bluetooth, so please be gentle.

Here's what I think is happening:

Initialize DFU process with only one device in range, and application is updated successfully. No problem.

Same body of code, initialize DFU process with multiple devices in range (not paired) and all devices enter DFU mode and only one is updated.

Using Xcode 8.3.2, with nrf52 based device, objective C application using the iOS Pods DFU Library, iOS 10.3.2 (although reports suggest any version of 10 and haven't yet tried back versions of iOS).

Also, none of the delegate methods provide a device parameter such that I can check whether responding device is targeted device.

I note that there is an issue requesting this as a feature ;-)

Also, using the nRF Toolbox, I do not see this problem.

Suggestions on how to debug and resolve?

Thanks!

Chris

philips77, 2017-06-12:
Hi,
Yes, we know it may be an issue. This may happen when you try to update a device in app mode and some other devices in range are already in DFU bootloader mode. This is because on iOS the device address is not available using standard CoreBluetooth API. Instead, a developer may define a DfuDeviceSelector that should point to the correct target based on its advertising data. By default it selects any device advertising with DFU service UUID.
After switching a selected target to bootloader mode the app tries to find it using this selector and connect to it. If it hasn't been bonded, the device will change its address by +1 so the same instance of CBPeripheral will not work. The lib will scan and determine the next target using the selector. But if you had more devices in DFU mode in range it will pick a random one, one that got picked up first.
This will be solved in SDK 14 where the app will request a specific data to be advertised when in DFU mode.

mcclen, 2017-06-12:
There would seem to be something else going on here which is the triggering condition. That is how does more than one device get into DFU boot loader mode. All devices in my scenario start in advertising mode and the process of putting the target device in DFU boot loader mode seems to cause all unpaired devices to go into DFU boot loader mode as well. How can I avoid the triggering condition?

philips77, 2017-06-15:
The trigger from app to BL mode is done only after the first connection after starting DFU. The following connection assumes you are in DFU state (it scans for DFU service UUID). Looks like you are starting DFU multiple times for each device that was found...? Please check your code.

mcclen, 2017-06-21:
This may be a forest for the trees issue - but I'm not seeing any explicit loop involving DFU within my code used to enter the DFU lib. I've compared to the example app and other code examples and don't "see" any difference of significance. I've changed a few things to more exactly match the example, but these have no effect on the available devices entering boot loader mode.

The target device is paired and bonded with service characteristics discovered and notifications set on non DFU services. Could that be a contributor?

mcclen, 2017-06-21:
Ta Da! While DFU is in progress, do not request scanning of central in other threads. This was my problem. The application is requesting device discovery. A device is selected for and discovered to required update. At this point the application needs to stop requesting device discovery.

Closing this item.

closed 2017-06-21 by mcclen