nordicsemi/IOS-DFU-Library #118: Please tell me how to manual installation the framework to my project?

developerXiong, 2017-06-14:
thanks , thanks , !!!!

mostafaberg, 2017-06-14:
Hi @developerXiong !
What do you mean by manual installation ?

developerXiong, 2017-06-14:
sorry,my English is poor, I mean install don't use cocoaPods nor carthage.

mostafaberg, 2017-06-14:
No problems at all, we currently don't officially support that because we would like to have a reproducible build in case of bugs, and Cocoapods / Carthage gives us that kind of certainty.

You can stil build your own by cloning the project onto your computer, then dragging the library into your own Xcode project

developerXiong, 2017-06-14:
@mostafaberg  thanks for your reply, the last question is gone, now I use carthage install like library [https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library](url), but when I use iOSDFULibrary my project is crash!

developerXiong, 2017-06-14:
and I try install other frameworks is OK. nothing the xcode tell me , if you have anything advice please tell me,thank you very much

mostafaberg, 2017-06-14:
Can you show us some code? and can you show us when it crashes what happens exactly ?

developerXiong, 2017-06-14:
`#import "DFUUpdateViewController.h"
#import <CoreBluetooth/CoreBluetooth.h>

//@import iOSDFULibrary;
#import <iOSDFULibrary/iOSDFULibrary-Swift.h>
//@import iOSDFULibrary;

@interface DFUUpdateViewController () <CBCentralManagerDelegate, CBPeripheralDelegate ,DFUServiceDelegate, DFUProgressDelegate, LoggerDelegate>

/**
 dfu controller
 */
@property (nonatomic, strong) DFUServiceController *dfuController;

@property (nonatomic, assign) BOOL secureDFU;

/**
 固件
 */
//@property (nonatomic, strong) DFUFirmware *selectedFirmware;

@property (nonatomic, strong) CBUUID *ExperimentalButtonlessDfuUUID;
@property (nonatomic, strong) CBUUID *legacyDfuServiceUUID;
@property (nonatomic, strong) CBUUID *secureDfuServiceUUID;
@property (nonatomic, strong) CBUUID *deviceInfoServiceUUID;

@end

@implementation DFUUpdateViewController

- (void)viewDidLoad {
    [super viewDidLoad];
    
    _manager.delegate = self;
    
    // 初始化uuid
    [self initUUID];
    // 连接dfu设备
    [self connectDFUPeripheral];

    
    UIButton *btn = [UIButton buttonWithType:UIButtonTypeCustom];
    [btn setFrame:CGRectMake(100, 200, 100, 44)];
    [btn setTitle:@"begin" forState:UIControlStateNormal];
    [btn setTitleColor:[UIColor redColor] forState:UIControlStateNormal];
    [btn addTarget:self action:@selector(startDFUProcess) forControlEvents:UIControlEventTouchUpInside];
    [self.view addSubview:btn];
    
}

- (void)initUUID {
    _ExperimentalButtonlessDfuUUID = [CBUUID UUIDWithString:@"8E400001-F315-4F60-9FB8-838830DAEA50"];
    _legacyDfuServiceUUID = [CBUUID UUIDWithString:@"00001530-1212-EFDE-1523-785FEABCD123"];
    _secureDfuServiceUUID = [CBUUID UUIDWithString:@"FE59"];
    _deviceInfoServiceUUID = [CBUUID UUIDWithString:@"180A"];
}

// 连接dfu设备
- (void)connectDFUPeripheral {
    [_manager connectPeripheral:_dfuPeripheral options:nil];
}

// 获取固件包的URL
- (NSURL *)getFirmwareURL {
    return [[NSBundle mainBundle] URLForResource:@"dfu-2" withExtension:@"zip"];
}

// 获取固件
- (DFUFirmware *)getDFUFirmware {
    return [[DFUFirmware alloc] initWithUrlToZipFile:[self getFirmwareURL]];
}

// 开始dfu升级
- (void)startDFUProcess {
    DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:_manager target:_dfuPeripheral];
    
    initiator.delegate = self;
    initiator.progressDelegate = self;
    initiator.logger = self;
    
    // This enables the experimental Buttonless DFU feature from SDK 12.
    // Please, read the field documentation before use.
    initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
    
    _dfuController = [[initiator withFirmware:[self getDFUFirmware]] start];
}

#pragma mark - DFUServiceDelegate

// dfu状态改变调用
- (void)dfuStateDidChangeTo:(enum DFUState)state {
    if (state == DFUStateCompleted || state == DFUStateDisconnecting) {
        // dfu升级完成或者失去连接
        
        
    } else if (state == DFUStateAborted) {
        // dfu升级中止
        
        
    } else {
        
        
        
    }
    
    // 升级完成之后将 DFUController置为空
    if (state == DFUStateCompleted) {
        _dfuController = nil;
    }
}

// 升级过程中报错
- (void)dfuError:(enum DFUError)error didOccurWithMessage:(NSString *)message {
    NSLog(@"升级过程中报错:%@ ", message);
}

#pragma mark - DFUProgressDelegate

- (void)dfuProgressDidChangeFor:(NSInteger)part outOf:(NSInteger)totalParts to:(NSInteger)progress currentSpeedBytesPerSecond:(double)currentSpeedBytesPerSecond avgSpeedBytesPerSecond:(double)avgSpeedBytesPerSecond {
    
    NSLog(@"部分：%ld/%ld, 进度:%ld%%, 速度：%.1fKB/s, 平均速度：%.1fKB/s", (long)part, (long)progress, (long)totalParts, currentSpeedBytesPerSecond/1024, avgSpeedBytesPerSecond/1024);
    
}

#pragma mark - LoggerDelegate

- (void)logWith:(enum LogLevel)level message:(NSString *)message {
//    NSLog(@"结束了？？？？？？%ld ---- %@", (long)level, message);
}

#pragma mark - cetralManager delegate

- (void)centralManagerDidUpdateState:(CBCentralManager *)central {
    
}

// 连接上dfu设备调用
- (void)centralManager:(CBCentralManager *)central didConnectPeripheral:(CBPeripheral *)peripheral {
    NSLog(@"连上dfu设备:%@", peripheral.name);
    peripheral.delegate = self;
    [peripheral discoverServices:nil];
}

// 失去dfu设备连接调用
- (void)centralManager:(CBCentralManager *)central didDisconnectPeripheral:(CBPeripheral *)peripheral error:(NSError *)error {
    
}

#pragma mark - peripheral delegate
// 发现设备的服务调用
- (void)peripheral:(CBPeripheral *)peripheral didDiscoverServices:(NSError *)error {
    NSArray *services = peripheral.services;
    for (CBService *service in services) {
        if ([service.UUID isEqual:_legacyDfuServiceUUID]) {
            _secureDFU = NO;
        } else if ([service.UUID isEqual:_secureDfuServiceUUID]) {
            _secureDFU = YES;
        } else if ([service.UUID isEqual:_ExperimentalButtonlessDfuUUID]) {
            _secureDFU = YES;
        } else {
            _secureDFU = NO;
        }
    }
    
}

- (void)touchesBegan:(NSSet<UITouch *> *)touches withEvent:(UIEvent *)event {
    [super touchesBegan:touches withEvent:event];
    
    [self dismissViewControllerAnimated:YES completion:nil];
}
`

developerXiong, 2017-06-14:
It's OK? when I click the begin button the project crash and nothing show me.

developerXiong, 2017-06-14:
[image]
I import the frameworks like this.

developerXiong, 2017-06-15:
Are you looking?

developerXiong, 2017-06-15:
Is it installed with Carthage and can only be run in the swift project?

developerXiong, 2017-06-15:
Are your libraries running through the Carthage and not running in the Objective-C project?

developerXiong, 2017-06-15:
Hah, it's okay to call libraries with swift code, but if you know why there is an error with Objective-C code calls, let me know why please, thank you very much!!!

mostafaberg, 2017-06-15:
The library works on all different installations through carthage, cocoapods with objc or swift, so that's not a problem.

I still don't know what's the exact error you get, can you show us that ?

developerXiong, 2017-06-15:
Well, I'm using the CocoaPods installed by iOSDFULibrary, and there's no problem running the project in swift and objc. Installing iOSDFULibrary using Carthage is no problem importing in swift, but the program that Carthage installs when iOSDFULibrary is imported in objc crashes when used.

developerXiong, 2017-06-15:
我用swift写的代码是没有问题的，但是用objc写的时候，程序会崩溃。。。

mostafaberg, 2017-06-15:
Can you show us exactly what happens ? what do you mean `crashes when used` specifically

developerXiong, 2017-06-15:
DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:_manager target:_dfuPeripheral];
    
    initiator.delegate = self;
    initiator.progressDelegate = self;
    initiator.logger = self;
    
    // This enables the experimental Buttonless DFU feature from SDK 12.
    // Please, read the field documentation before use.
    initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
    
    _dfuController = [[initiator withFirmware:[self getDFUFirmware]] start];

mostafaberg, 2017-06-15:
This is your code, not the `crash`, I need to see what the actual crash is so i can know what is wrong

developerXiong, 2017-06-15:
there's have anything about crash info

developerXiong, 2017-06-15:
There is no information about crashes

mostafaberg, 2017-06-15:
I'm sorry but It's really hard to help you without knowing what crashes, I don't see anything that is wrong with your code.

When you say crashes, what do you mean exactly by that ? what crashes and how?

developerXiong, 2017-06-15:
if you have time, you can try like me, maybe you can find the problem.

developerXiong, 2017-06-15:
I'm sorry to have delayed you for such a little oversight that the problem has been settled.

closed 2017-06-15 by developerXiong

mostafaberg, 2017-06-15:
Good to hear you got it working!, Sorry we couldn't help so much without more information.
what was the problem after all ?