nordicsemi/IOS-DFU-Library #119: Problems compiling

glawe, 2017-06-14:
Hi,
I tried to get the library working inside a new objective-ca app project using xcode 8.3.3. 
I followed the instructions here and set it up as a pod:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/README_OBJC.md

However, when I try to compile the project swift crashes with the following message:

Any idea of what it could be ?

Users/fredrik/Library/Developer/Xcode/DerivedData/......../Build/Intermediates/Pods.build/Debug-iphoneos/iOSDFULibrary.build/Objects-normal/arm64/DFUPeripheral.swiftdeps:318:29: error: Could not find closing ]!
- !private ["CSo9CBManager", 
                            ^
0  swiftc                   0x000000010f64b4f7 PrintStackTraceSignalHandler(void*) + 39
1  swiftc                   0x000000010f64a9a6 SignalHandler(int) + 646
2  libsystem_platform.dylib 0x00007fff9afefb3a _sigtramp + 26
3  libsystem_platform.dylib 0x0000000000000004 _sigtramp + 1694565604
4  swiftc                   0x000000010c42735f swift::driver::Compilation::performJobsImpl() + 1727
5  swiftc                   0x000000010c41b4b4 main + 19108
6  libdyld.dylib            0x00007fff9ade0235 start + 1
7  libdyld.dylib            0x0000000000000079 start + 1696726597
Stack dump:
0.	Program arguments: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc -

mostafaberg, 2017-06-14:
Have you tried clearing your derived data directory?

glawe, 2017-06-14:
That did the trick.
Cheers! ;)

mostafaberg, 2017-06-14:
Good to hear :)

closed 2017-06-14 by mostafaberg