nordicsemi/IOS-DFU-Library #120: Error 304: Enabling notifications failed

subhash08, 2017-06-14:
When I'm trying to Firmware updation , it's not working, here is log = 

<CBPeripheral: 0x1740f5a00, identifier = 907980BC-8FC6-4675-8A2A-F2EFAB1A16EC, name = SIDEKICK, state = connected>
Changed state to: Connecting
I : Connected to SIDEKICK
I : Services discovered
V : Starting Secure DFU...
I : Connected to SIDEKICK
I : Services discovered
V : Secure DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, for: FE59)
I : DFU characteristics discovered
V : Enabling notifications for 8E400001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.setNotifyValue(true, for: 8E400001-F315-4F60-9FB8-838830DAEA50)
Changed state to: Starting
E : Enabling notifications failed
E : Error 3: Writing is not permitted.
V : Disconnecting...
D : centralManager.cancelPeripheralConnection(peripheral)
D : [Callback] Central Manager did disconnect peripheral
I : Disconnected
Error 304: Enabling notifications failed

Please let me Know what was the issue
`func startDFUProcess() {
        guard dfuPeripheral != nil else {
            print("No DFU peripheral was set")
            return
        }
        
        let dfuInitiator:DFUServiceInitiator = DFUServiceInitiator(centralManager: appDel.manager, target: dfuPeripheral!).with(firmware: selectedFirmware!)
        dfuInitiator.delegate = self
        dfuInitiator.progressDelegate = self
        dfuInitiator.logger = self
        
       dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
        
        dfuController = dfuInitiator.start()
    }`

philips77, 2017-06-15:
Hi, it looks like a caching problem. The lib tries to enable notifications but it writes to an invalid handle and writing not permitted error is returned. 
If your device is not bonded, make sure you have the Service Changed characteristic in the Generic Attribute service, this way every time you connect to it the service discovery will be performed.
If your device is bonded you have to rely on Service Change indication whenever your services have changed. To clear the cache remove bond and restart Bluetooth.

subhash08, 2017-06-15:
Hello, I tried to reconnect BLE and restart it, but not working.
Device have Service Changed characteristic in the Generic Attribute service.
Its updating from android app using your Lib but in iOS it isn't working. 
Help me to figure it out, what is the issue.

subhash08, 2017-06-16:
Hello, any suggestion and help please.

mostafaberg, 2017-06-16:
Are you bonded to the device ?

subhash08, 2017-06-16:
Yes, device is bonded.

subhash08, 2017-06-16:
With other device , try to Firmware updation
<CBPeripheral: 0x1740f7b00, identifier = 85BF0552-AB4D-4B52-B039-560C9881BD32, name = ASTER, state = connected>
Changed state to: Connecting
I : Connected to ASTER
I : Services discovered
V : Starting Secure DFU...
I : Connected to ASTER
I : Services discovered
V : Secure DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, for: FE59)
I : DFU characteristics discovered
V : Enabling notifications for 8E400001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.setNotifyValue(true, for: 8E400001-F315-4F60-9FB8-838830DAEA50)
Changed state to: Starting
V : Notifications enabled for 8E400001-F315-4F60-9FB8-838830DAEA50
A : Buttonless DFU notifications enabled
W : Application with buttonless update found
V : Writing to characteristic 8E400001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x01, for: 8E400001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Changed state to: Enabling DFU Mode
I : Data written to 8E400001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8E400001-F315-4F60-9FB8-838830DAEA50, value (0x):200101
A : Response (Op Code = 1, Status = 1) received

after it device disconnected and nothing happens.
Please help me.

mostafaberg, 2017-06-16:
Interesting!, are you sure your device actually indicates a service change? that might be the issue!
do you have a sniffer ?

subhash08, 2017-06-16:
@mostafaberg  I'm sure my device  have a service change, It's working perfectly with Android app,
Its only happening in iOS.

mostafaberg, 2017-06-16:
Android doesn't rely on service changed indications so it'll work both ways, that's why i'm asking if you have a sniffer to see what's actually going on, which version of iOS and version of the library is that issue happening on ?

subhash08, 2017-06-16:
Sorry, I don't have a sniffer, I'm using iOS 10.3.2 n library version 3.1.0.

mostafaberg, 2017-06-16:
Do you have access to another iOS device with a different iOS version ? I want to rule out that this is not a bug with iOS, if not then perhaps your peripheral is not sending service changed indications, that's my closest guess, but unfortunately you can't compare the android implementation as it's completely different than the iOS one.

subhash08, 2017-06-16:
I have another device iPhone7 with iOS 11 beta and  iphone 5s , iPad with iOS 10.3.1. let me check it on other device.

mostafaberg, 2017-06-16:
Sounds perfect!, let me know how it goes!

subhash08, 2017-06-16:
In iphone 5 iOS 10.3.1 , 
when try to update firmware, logs 

<```
CBPeripheral: 0x1756aef0, identifier = 90C91A80-EC5E-4E73-BA4D-64AC97062F51, name = SIDEKICK, state = connected>
I : Connected to SIDEKICK
Changed state to: Connecting
I : Services discovered
V : Starting Secure DFU...
I : Connected to SIDEKICK
I : Services discovered
V : Secure DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, for: FE59)
I : DFU characteristics discovered
V : Enabling notifications for 8E400001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.setNotifyValue(true, for: 8E400001-F315-4F60-9FB8-838830DAEA50)
Changed state to: Starting
V : Notifications enabled for 8E400001-F315-4F60-9FB8-838830DAEA50
A : Buttonless DFU notifications enabled
W : Application with buttonless update found
V : Writing to characteristic 8E400001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x01, for: 8E400001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Changed state to: Enabling DFU Mode
I : Data written to 8E400001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8E400001-F315-4F60-9FB8-838830DAEA50, value (0x):200101
A : Response (Op Code = 1, Status = 1) received
D : [Callback] Central Manager did disconnect peripheral
I : Disconnected by the remote device
V : Scanning for the DFU Bootloader...
I : DFU Bootloader found with name Sidekick_dfu
V : Connecting to Sidekick_dfu...
D : centralManager.connect(peripheral, options: nil)
D : [Callback] Central Manager did connect peripheral
I : Connected to Sidekick_dfu
V : Discovering services...
D : peripheral.discoverServices([FE59])
I : Services discovered
V : Secure DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, for: FE59)
I : DFU characteristics discovered
V : Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
Changed state to: Starting
V : Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
A : Secure DFU Control Point notifications enabled
V : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
A : Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
V : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x010190000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
V : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
A : Packet Receipt Notif disabled (Op Code = 2, Value = 0)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x128d010a470801124308ffffffff0f10ffffffff, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x0f1a03feff0320002800300038908a0342240803, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x1220d99f58526724999b5bdfc8bf44d35c1dcf64, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0xf2f93bd50d11d876a83772417ee6480110001a40, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0xdd37e51042e03cee43e0077392122ab5847d1058, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0xb0ab83e57e0786e9a91bc93a56857a6404483e13, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0xa2c28736e5cc00239c2f51862cca771630974727, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V : Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0xccc471c0, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
A : Command object sent (CRC = 26475E0C)
V : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600301900000000c5e4726
A : Checksum (Offset = 144, CRC = 26475E0C) received
V : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D : peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600405
E : Error 5: Invalid object
V : Disconnecting...
D : centralManager.cancelPeripheralConnection(peripheral)
D : [Callback] Central Manager did disconnect peripheral
I : Disconnected
Error 15: Invalid object
```

and device disconnected , then nothing happens

subhash08, 2017-06-19:
Any help brother, still not able to figure out the issue?

mostafaberg, 2017-06-19:
From what i'm seeing, the other phone flashed your device but the firmware was invalid as you see in this line:
```
E : Error 5: Invalid object
```

Since the issue isn't happening then it was probably a caching problem. you are now having another problem with the invalid object part and this you have to try to fix with your firmware team, could be many reasons so it's hard to tell without knowing your hardware and firmware compatibility, might also be the way you generate your firmware package.

for now , i'll close this issue as your main problem is resolved, if you require help with the other firmware, please post it on our [devzone](http://devzone.nordicsemi.com) and not here as it's not an issue with our library. remember to post as much information and logs as possible on the Devzone so our engineers can help you as fast as possible

closed 2017-06-19 by mostafaberg

subhash08, 2017-06-19:
Thanks @mostafaberg  for your  patience
The main issue of  "Error 304: Enabling notifications failed" was resolved when I update DFU library to current version. 
 other issue was with my firmware file.
Thank you man.

mostafaberg, 2017-06-19:
No worries, glad to hear it worked out :)

vijmax, 2021-11-25:
Hi, iam using the latest SDK and iam still facing the same problem. Couple of observations ...
1. Rebooting the phone does the trick. Iam able to upgrade. But it allows only once, after that iam stuck with error 304 again.
2. Same problem seen with stock nRF Toolbox from App store as well