nordicsemi/IOS-DFU-Library #121: Change DFU Service UUID

rahul-sciter, 2017-06-15:
I have multiple bluetooth devices which have different DFU Service UUID. How can i change DFU Service UUID runtime using iOS DFU Library.

philips77, 2017-06-15:
Hi, currently it is not possible in iOS DFU library. May I ask why did you decide to have multiple UUIDs for DFU?

rahul-sciter, 2017-06-15:
We have developed multiple hardware each have different UUID for DFU Service. So I have developed one application which support all bluetooth devices. So I want to integrate firmware upgrade for all bluetooth devices. So if user select device for firmware upgrade based on that i need to scan DFU Device.

rahul-sciter, 2017-06-15:
In android library you have provided one flag EXTRA_CUSTOM_UUIDS_FOR_LEGACY_DFU on that bases your android library works for multiple DFU Service UUID.

rahul-sciter, 2017-06-15:
In Android library in this class **UuidHelper.java** there is one function **UuidHelper.assignCustomUuids(intent);** by calling it we can assign UUID from our application.
Is this possible in iOS Library?

rahul-sciter, 2017-06-16:
Can anyone answer me on this?

mostafaberg, 2017-06-16:
@rahul-sciter Sorry for the late response, right now it's not officially supported on the iOS library, but if you want to [overwrite it yourself](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L26) that should be ok, if you want to modify it at runtime it's just a matter of adding your own initializer that takes in a custom `UUID` for the service, it's hard to tell the exact solution without seeing your use case and code, but that should be the way to do it

mostafaberg, 2017-07-21:
@rahul-sciter Any updates on the issue ? have you got things working or do you require help ?

closed 2017-08-08 by mostafaberg