nordicsemi/IOS-DFU-Library #125: I need Object-C Library.

YoonYoungDoo, 2017-07-21:
Hello.
We sell device using nrf52832 chip in Korea. 
Unfortunately, We don't use cocoapods in App. So, I'm had change this library to Framework.
And that framework use in my application building Object-C. This is perfect! All function is nomal.
but, that framework have big size. iOSDFULibrary.framework have 25MB.
I am wondering if I should create a new library with Object-C.
please help me. Thank.

mostafaberg, 2017-07-21:
Good morning @YoonYoungDoo !
How do you generate the iOSDFULibrary.framework ? can you attach the said framework file so we can analyze it ? my guess is that the size is due to the swift standard library and having multiple build targets, what you should test is your final project size when downloaded from the app store.
but still you can send us the actual framework and we can take it from there. thanks

YoonYoungDoo, 2017-07-24:
I build this library with release. that have armv7, armv7s, arm64.
And this size is nomal. Maybe 2.5MB?
But if build my app using this framework, app size is big.
I guess that auto add libSwift.
This is size.

iOSDFULibrary.framework  1.6MB
libswiftCore.dylib                 14.7MB
libswiftCoreGraphics.dylib  545KB
libswiftCoreImage.dylib       208KB
libswiftDarwin.dylib              297KB
libswiftDispatch.dylib          1.1MB
libswiftFoundation.dylib      5.7MB
libswiftObjectiveC.dylib      259KB
libswiftQuartzCore.dylib     213KB
libswiftUIKit.dylib                354KB

YoonYoungDoo, 2017-08-03:
[http://ehisoft.co.kr/public/ios_dfu_framework.zip](url)
This is Framework. Please check.

mostafaberg, 2017-08-03:
Nothing seems wrong to me, you are assuming that this size is the one that your users will be downloading from the app store, which is not the case, app store distributions will be much smaller in size depending on the device that downloads them, you might want to read a bit about [bitcode and app thinning](https://developer.apple.com/library/content/documentation/IDEs/Conceptual/AppDistributionGuide/AppThinning/AppThinning.html#//apple_ref/doc/uid/TP40012582-CH35-SW3)

I would say just ignore the big files on export and check the actual download sizes on your production releases. so this is not really an issue with our library or the way you build it

closed 2017-08-03 by mostafaberg