nordicsemi/IOS-DFU-Library #127: About BLE pairing on the IOS platform...

developerXiong, 2017-08-11:
I have a question to ask you about Bluetooth pairing.I want to use the code to enter the pairing password and cancel the system pop-up window when it is paired. if you have answer, please tell me, thank you very much!!!!!!

mostafaberg, 2017-08-11:
Hi @developerXiong Pairing on iOS is handled by the operating system and is not overridable by the developers, so you will have to use the system popup unfortunately

closed 2017-08-11 by mostafaberg

developerXiong, 2017-08-11:
if I add the method  - (void)peripheral:(CBPeripheral *)peripheral didWriteValueForCharacteristic:(CBCharacteristic *)characteristic error:(nullable NSError *)error,,, and write here nothing, the system pop-up will be hide...

mostafaberg, 2017-08-11:
Must be something else that's crashing, but anyways no matter what you attempt you can't override that iOS has to handle the Pairing and not your app

developerXiong, 2017-08-11:
OK! thank you