nordicsemi/IOS-DFU-Library #129: Error 202: Device disconnected unexpectedly

pjl1234567, 2017-08-25:
This error is always present：Error 202: Device disconnected unexpectedly

2017-08-25 15:58:11.822899+0800[1146:324159] [✅][Framework][5]:Connected to ShanrenD
2017-08-25 15:58:11.823091+0800[1146:324159] [✅][Framework][1]:Discovering services...
2017-08-25 15:58:11.823261+0800[1146:324159] [✅][Framework][0]:peripheral.discoverServices(nil)
2017-08-25 15:58:11.919692+0800[1146:324159] [✅][Framework][0]:[Callback] Central Manager did connect peripheral
2017-08-25 15:58:11.919867+0800[1146:324159] [✅][Framework][5]:Connected to ShanrenDFU
2017-08-25 15:58:11.919969+0800[1146:324159] [✅][Framework][1]:Discovering services...
2017-08-25 15:58:11.920062+0800[1146:324159] [✅][Framework][0]:peripheral.discoverServices(nil)
2017-08-25 15:58:12.292954+0800[1146:324159] [✅][Framework][5]:Services discovered
2017-08-25 15:58:12.293253+0800[1146:324159] [✅][Framework][1]:Starting Legacy DFU...
2017-08-25 15:58:12.293587+0800[1146:324159] [✅][Framework][5]:Connected to ShanrenDFU
2017-08-25 15:58:12.293793+0800[1146:324159] [✅][Framework][5]:Services discovered
2017-08-25 15:58:12.293967+0800[1146:324159] [✅][Framework][1]:Legacy DFU Service found
2017-08-25 15:58:12.294175+0800[1146:324159] [✅][Framework][1]:Discovering characteristics in DFU Service...
2017-08-25 15:58:12.294378+0800[1146:324159] [✅][Framework][0]:peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
2017-08-25 15:58:12.534498+0800[1146:324159] [✅][Framework][5]:DFU characteristics discovered
2017-08-25 15:58:12.534870+0800[1146:324159] [✅][Framework][1]:Reading DFU Version number...
2017-08-25 15:58:12.535090+0800[1146:324159] [✅][Framework][0]:peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
2017-08-25 15:58:12.592306+0800[1146:324159] [✅][Framework][5]:Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
2017-08-25 15:58:12.592569+0800[1146:324159] [✅][Framework][10]:Version number read: 0.8
2017-08-25 15:58:12.592883+0800[1146:324159] [✅][Framework][1]:Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
2017-08-25 15:58:12.593678+0800[1146:324159] [✅][Framework][0]:peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
2017-08-25 15:58:12.741258+0800[1146:324159] [✅][Framework][1]:Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
2017-08-25 15:58:12.741442+0800[1146:324159] [✅][Framework][10]:DFU Control Point notifications enabled
2017-08-25 15:58:12.741692+0800[1146:324159] [✅][Framework][1]:Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-08-25 15:58:12.741919+0800[1146:324159] [✅][Framework][0]:peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-08-25 15:58:12.743284+0800[1146:324159] [✅][Framework][1]:Writing image sizes (0b, 0b, 26584b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
2017-08-25 15:58:12.743922+0800[1146:324159] [✅][Framework][0]:peripheral.writeValue(0x0000000000000000d8670000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
2017-08-25 15:58:12.801126+0800[1146:324159] [✅][Framework][5]:Data written to 00001531-1212-EFDE-1523-785FEABCD123
2017-08-25 15:58:12.801312+0800[1146:324159] [✅][Framework][10]:Start DFU (Op Code = 1, Upload Mode = 4) request sent
2017-08-25 15:58:14.992830+0800[1146:324159] [✅][Framework][5]:Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
2017-08-25 15:58:14.993141+0800[1146:324159] [✅][Framework][10]:Response (Op Code = 1, Status = 1) received
2017-08-25 15:58:14.993421+0800[1146:324159] [✅][Framework][10]:Writing Initialize DFU Parameters...
2017-08-25 15:58:14.993631+0800[1146:324159] [✅][Framework][1]:Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-08-25 15:58:14.994039+0800[1146:324159] [✅][Framework][0]:peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-08-25 15:58:14.995001+0800[1146:324159] [✅][Framework][1]:Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
2017-08-25 15:58:14.995761+0800[1146:324159] [✅][Framework][0]:peripheral.writeValue(0xff00ff00ffffffff01006400e7bd, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
2017-08-25 15:58:15.001336+0800[1146:324159] [✅][Framework][1]:Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2017-08-25 15:58:15.001681+0800[1146:324159] [✅][Framework][0]:peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2017-08-25 15:58:15.082061+0800[1146:324159] [✅][Framework][5]:Data written to 00001531-1212-EFDE-1523-785FEABCD123
2017-08-25 15:58:15.132712+0800[1146:324159] [✅][Framework][0]:[Callback] Central Manager did disconnect peripheral
2017-08-25 15:58:15.137018+0800[1146:324159] [✅][Framework][5]:Disconnected by the remote device
2017-08-25 15:58:15.138383+0800[1146:324159] Error 202: Device disconnected unexpectedly
2017-08-25 15:58:15.139918+0800[1146:324159] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x17487a8c0> can only accept this command while in the powered on state
2017-08-25 15:58:15.151315+0800[1146:324159] -[BLEManager centralManagerDidUpdateState:] [Line 83] CBPeripheralManagerStatePoweredOn

mostafaberg, 2017-08-25:
Seems like your Bluetooth is getting turned off for some reason ?
```
2017-08-25 15:58:15.139918+0800[1146:324159] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x17487a8c0> can only accept this command while in the powered on state
2017-08-25 15:58:15.151315+0800[1146:324159] -[BLEManager centralManagerDidUpdateState:] [Line 83] CBPeripheralManagerStatePoweredOn
```

What phone are you using and what version of iOS ? did you try restarting the phone ?

mostafaberg, 2017-08-29:
Closing the issue during the week due to inactivity, if you have an update and the issue is closed please leave a comment and I'll reopen :)

gkovelman, 2017-09-11:
I'm experiencing the same behavior.
I've recently updated the DFU library from 3.0.6 to 3.2.0 and this error now occurs when I try to update a device with SDK 8.0. With DFU 3.0.6, the procedure worked well.

Here's more details:
iOS: 10.3.2
Device: iPhone 6
NRF iOS SDK version: 3.2.0

DFU Service Initiator parameters:
force_dfu = false
packetReceiptNotificationParameter = 12
enableUnsafeExperimentalButtonlessServiceInSecureDfu = true (didn't work with 'false' as well)

NRF LOG:
RF DFU Log (2017-09-11 14:17:26 +0000) : Connecting to Dfu52E5...
NRF DFU Log (2017-09-11 14:17:26 +0000) : centralManager.connect(peripheral, options: nil)
NRF DFU Log (2017-09-11 14:17:26 +0000) : [Callback] Central Manager did connect peripheral
NRF DFU Log (2017-09-11 14:17:26 +0000) : Connected to Dfu52E5
NRF DFU Log (2017-09-11 14:17:26 +0000) : Discovering services...
NRF DFU Log (2017-09-11 14:17:26 +0000) : peripheral.discoverServices(nil)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Services discovered
NRF DFU Log (2017-09-11 14:17:27 +0000) : Starting Legacy DFU...
NRF DFU Log (2017-09-11 14:17:27 +0000) : Connected to Dfu52E5
NRF DFU Log (2017-09-11 14:17:27 +0000) : Services discovered
NRF DFU Log (2017-09-11 14:17:27 +0000) : Legacy DFU Service found
NRF DFU Log (2017-09-11 14:17:27 +0000) : Discovering characteristics in DFU Service...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
NRF DFU Log (2017-09-11 14:17:27 +0000) : DFU characteristics discovered
NRF DFU Log (2017-09-11 14:17:27 +0000) : Reading DFU Version number...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
NRF DFU Log (2017-09-11 14:17:27 +0000) : Version number read: 0.8
NRF DFU Log (2017-09-11 14:17:27 +0000) : Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:27 +0000) : DFU Control Point notifications enabled
NRF DFU Log (2017-09-11 14:17:27 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Writing image sizes (0b, 0b, 38252b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.writeValue(0x00000000000000006c950000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:27 +0000) : Start DFU (Op Code = 1, Upload Mode = 4) request sent
NRF DFU Log (2017-09-11 14:17:29 +0000) : Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
NRF DFU Log (2017-09-11 14:17:29 +0000) : Response (Op Code = 1, Status = 1) received
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing Initialize DFU Parameters...
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:29 +0000) : peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:29 +0000) : peripheral.writeValue(0xffffffffffffffff0100feffc146, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:29 +0000) : peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:29 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
NRF DFU Log (2017-09-11 14:17:30 +0000) : Response (Op Code = 2, Status = 1) received
NRF DFU Log (2017-09-11 14:17:30 +0000) : Initialize DFU Parameters completed
NRF DFU Log (2017-09-11 14:17:30 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:30 +0000) : peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:30 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
NRF DFU Log (2017-09-11 14:17:30 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:30 +0000) : peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:30 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Uploading firmware...
NRF DFU Log (2017-09-11 14:17:30 +0000) : Sending firmware to DFU Packet characteristic...
NRF DFU Log (2017-09-11 14:17:43 +0000) : [Callback] Central Manager did disconnect peripheral
NRF DFU Log (2017-09-11 14:17:43 +0000) : Disconnected by the remote device
NRF DFU ERROR DFUError : Device disconnected unexpectedly

closed 2017-09-22 by mostafaberg