nordicsemi/IOS-DFU-Library #135: I don't think it's accurate.

developerXiong, 2017-09-12:
[image]

I can use Carthage too.

and I have another problem ,when I do everything follow the picture ,the Xcode back to me the error:

dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /private/var/containers/Bundle/Application/3169ED4D-6B60-4898-B9D5-037AB3C72039/CYBluetoothManager.app/Frameworks/Zip.framework/Zip
  Reason: no suitable image found.  Did find:
	/private/var/containers/Bundle/Application/3169ED4D-6B60-4898-B9D5-037AB3C72039/CYBluetoothManager.app/Frameworks/Zip.framework/Frameworks/libswiftCore.dylib: code signature invalid for '/private/var/containers/Bundle/Application/3169ED4D-6B60-4898-B9D5-037AB3C72039/CYBluetoothManager.app/Frameworks/Zip.framework/Frameworks/libswiftCore.dylib'

developerXiong, 2017-09-12:
Sorry, the problem has been solved.

mostafaberg, 2017-09-12:
Carthage is supported yes , but not tested as we mentioned, that's why things might not work out of the box as the issue you are having.

your issue is here:
```
code signature invalid for '/private/var/containers/Bundle/Application/3169ED4D-6B60-4898-B9D5-037AB3C72039/CYBluetoothManager.app/Frameworks/Zip.framework/Frameworks/libswiftCore.dylib'
```

We are not sure how to help with that part but I guess you need to check if Xcode is actually signing the bundled frameworks

closed 2017-09-12 by mostafaberg