nordicsemi/IOS-DFU-Library #136: initWithUrlToZipFile crash!

PansGitHub, 2017-09-12:
hello, IOS-Pods-DFU-Library works fine before，but today i got error with method  initWithUrlToZipFile,i am sure the firmwarefilePath is right and the firmware is valid(I test it at nrf-toolbox,work fine), can you help me how to resolve this problem? 

crash info :         Zip.swift           line-57
```
    /**
     Set of vaild file extensions
     */
    internal static var customFileExtensions: Set<String> = []
    error:thread1 signal-SIGABART
```

my code :
```
- (void)startDFUWithDfuTarg:(CBPeripheral *)dfuTarg {
    NSMutableString * filePath = [[NSSearchPathForDirectoriesInDomains(NSDocumentDirectory, NSUserDomainMask, YES) objectAtIndex:0] mutableCopy];
    [filePath appendString:[NSString stringWithFormat:@"/package/%@", [ConfigService shareService].packageName]];
    DFUFirmware * selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[NSURL fileURLWithPath:filePath]];
    DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:[PHBluetooth sharePHBluetooth].DFUCentralManager target:dfuTarg];
    [initiator withFirmware:selectedFirmware];
    initiator.forceDfu = YES;
    initiator.packetReceiptNotificationParameter = 12;
    initiator.delegate = self;
    initiator.progressDelegate = self;
    self.controller = [initiator start];
}
```

my cartfile.resolved
```
github "NordicSemiconductor/IOS-Pods-DFU-Library" "3.0.6"
github "ReactiveCocoa/ReactiveObjC" "2.1.2"
github "SnapKit/Masonry" "v1.0.2"
github "realm/realm-cocoa" "v2.4.3"
```

mostafaberg, 2017-09-12:
Can you first try to use DFU library version `3.2.1` and see if the issue is still there in the latest version ?

mostafaberg, 2017-09-12:
Also what is the full name of your zip file ?

PansGitHub, 2017-09-12:
thanks a lot! DFU library work fine now after updated to version 3.2.1

mostafaberg, 2017-09-12:
Perfect ! that's great news, keep up the good work 👍

closed 2017-09-12 by mostafaberg