nordicsemi/IOS-DFU-Library #137: I'm sorry I have so much problem...

developerXiong, 2017-09-12:
I want to commit the code to Cocoapods, An error has occurred:
- WARN  | xcodebuild:  iOSDFULibrary/iOSDFULibrary/Classes/Implementation/DFUPeripheralSelector.swift:33:16: warning: 'select(_:advertisementData:RSSI:)' is deprecated

mostafaberg, 2017-09-12:
No problems at all :)
Yes we are deprecating this method, you should use the new Select method instead, if you are not using this method at all, you don't need to do anything just use `allow-warnings` in your pod commands to bypass that, and we will eventually phase it out in a later release :)

developerXiong, 2017-09-12:
I have used the allow-warnings command, but that mistake still exists:(

mostafaberg, 2017-09-12:
What are you doing exactly ? can you show the commands you are using ?

developerXiong, 2017-09-12:
pod trunk push MyProjectName.podspec --allow-warnings

mostafaberg, 2017-09-12:
What version of cocoapods are you using ? `pod --version`

developerXiong, 2017-09-12:
1.1.1

mostafaberg, 2017-09-12:
first i would recommend using the latest version of cocoapods, second i think the problem is the ordering of the commands, I think this should work instead:
`pod --allow-warnings trunk push MyProjectName.podspec`

closed 2017-09-12 by mostafaberg

developerXiong, 2017-09-12:
It's OK, you are best man, thank you!

mostafaberg, 2017-09-12:
great to hear you got things working :) good luck!