nordicsemi/IOS-DFU-Library #140: Upgrade failed

developerXiong, 2017-09-14:
This is log, I don't quite understand:
1---Connecting to cooya ota...
2017-09-14 14:27:35.510895+0800 QmjkSDKDemo[1169:583687] 0---centralManager.connect(peripheral, options: nil)
2017-09-14 14:27:35.714494+0800 QmjkSDKDemo[1169:583687] 0---[Callback] Central Manager did connect peripheral
2017-09-14 14:27:35.714764+0800 QmjkSDKDemo[1169:583687] 5---Connected to cooya ota
2017-09-14 14:27:35.714998+0800 QmjkSDKDemo[1169:583687] 1---Discovering services...
2017-09-14 14:27:35.715137+0800 QmjkSDKDemo[1169:583687] 0---peripheral.discoverServices(nil)
2017-09-14 14:27:35.964554+0800 QmjkSDKDemo[1169:583687] 5---Services discovered
2017-09-14 14:27:35.967955+0800 QmjkSDKDemo[1169:583687] 1---Starting Secure DFU...
2017-09-14 14:27:35.969137+0800 QmjkSDKDemo[1169:583687] 5---Connected to cooya ota
2017-09-14 14:27:35.969330+0800 QmjkSDKDemo[1169:583687] 5---Services discovered
2017-09-14 14:27:35.969472+0800 QmjkSDKDemo[1169:583687] 1---Secure DFU Service found
2017-09-14 14:27:35.969831+0800 QmjkSDKDemo[1169:583687] 1---Discovering characteristics in DFU Service...
2017-09-14 14:27:35.970025+0800 QmjkSDKDemo[1169:583687] 0---peripheral.discoverCharacteristics(nil, for: FE59)
2017-09-14 14:27:35.973687+0800 QmjkSDKDemo[1169:583687] 5---DFU characteristics discovered
2017-09-14 14:27:35.975415+0800 QmjkSDKDemo[1169:583687] 1---Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:35.975613+0800 QmjkSDKDemo[1169:583687] 0---peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2017-09-14 14:27:36.083918+0800 QmjkSDKDemo[1169:583687] 1---Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2017-09-14 14:27:36.084170+0800 QmjkSDKDemo[1169:583687] 10---Secure DFU Control Point notifications enabled
2017-09-14 14:27:36.084681+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.085714+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2017-09-14 14:27:36.204067+0800 QmjkSDKDemo[1169:583687] 5---Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2017-09-14 14:27:36.205811+0800 QmjkSDKDemo[1169:583687] 5---Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
2017-09-14 14:27:36.206499+0800 QmjkSDKDemo[1169:583687] 10---Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
2017-09-14 14:27:36.206785+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.207643+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x010187000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2017-09-14 14:27:36.293526+0800 QmjkSDKDemo[1169:583687] 5---Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2017-09-14 14:27:36.353902+0800 QmjkSDKDemo[1169:583687] 5---Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2017-09-14 14:27:36.354252+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.354542+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2017-09-14 14:27:36.473542+0800 QmjkSDKDemo[1169:583687] 5---Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2017-09-14 14:27:36.474963+0800 QmjkSDKDemo[1169:583687] 5---Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
2017-09-14 14:27:36.475155+0800 QmjkSDKDemo[1169:583687] 10---Packet Receipt Notif disabled (Op Code = 2, Value = 0)
2017-09-14 14:27:36.475374+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.476072+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x1284010a3e0801123a080410331a028701200028, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2017-09-14 14:27:36.476935+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.477658+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x00300038c0b3034224080312206087c3a8e5cca5, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)

2017-09-14 14:27:36.479748+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.480472+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x8872b5e42ada67705e38405e8d55d104658d6c6c, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2017-09-14 14:27:36.481056+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.481745+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0xef3715a7be480010001a40b9c0c3848c88e65811, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2017-09-14 14:27:36.482310+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.482998+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0xa02d3547530596db0ae6dbf5a45c0676ff0cc981, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2017-09-14 14:27:36.483554+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.490960+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x9915f2aaa76396c3c61da87b64600c04a6d516b5, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2017-09-14 14:27:36.491736+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.492299+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0xc9d3c6e850119a27419b47de41d5e5, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2017-09-14 14:27:36.492992+0800 QmjkSDKDemo[1169:583687] 10---Command object sent (CRC = 8A238607)
2017-09-14 14:27:36.493265+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.493479+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2017-09-14 14:27:36.593727+0800 QmjkSDKDemo[1169:583687] 5---Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2017-09-14 14:27:36.595419+0800 QmjkSDKDemo[1169:583687] 5---Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600301870000000786238a
2017-09-14 14:27:36.595706+0800 QmjkSDKDemo[1169:583687] 10---Checksum (Offset = 135, CRC = 8A238607) received
2017-09-14 14:27:36.596035+0800 QmjkSDKDemo[1169:583687] 1---Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2017-09-14 14:27:36.596254+0800 QmjkSDKDemo[1169:583687] 0---peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2017-09-14 14:27:36.653494+0800 QmjkSDKDemo[1169:583687] 5---Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2017-09-14 14:27:36.713843+0800 QmjkSDKDemo[1169:583687] 5---Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60040a
2017-09-14 14:27:36.714116+0800 QmjkSDKDemo[1169:583687] 20---Error 10: Operation failed
2017-09-14 14:27:36.714264+0800 QmjkSDKDemo[1169:583687] 1---Disconnecting...
2017-09-14 14:27:36.714402+0800 QmjkSDKDemo[1169:583687] 0---centralManager.cancelPeripheralConnection(peripheral)
2017-09-14 14:27:36.717890+0800 QmjkSDKDemo[1169:583687] 0---[Callback] Central Manager did disconnect peripheral
2017-09-14 14:27:36.718081+0800 QmjkSDKDemo[1169:583687] 5---Disconnected

mostafaberg, 2017-09-14:
Thanks for the report !, have you tried setting your PRN value to 12 ? which version of the library is that ?

developerXiong, 2017-09-14:
Thanks for the answer, the problem is solved, because of the hardware permissions.

closed 2017-09-14 by developerXiong

mostafaberg, 2017-09-14:
Well that was easy ;) good luck further

madebyfireneil, 2017-12-04:
@developerXiong what did you ave to do, to change hardware permissions?