nordicsemi/IOS-DFU-Library #141: Error 202: Device disconnected unexpectedly after library upgrade

gkovelman, 2017-09-18:
Following issue #129, I'm experiencing the same behavior.
I've recently updated the DFU library from 3.0.6 to 3.2.0 and this error now occurs when I try to update a device with SDK 8.0. With DFU 3.0.6, the procedure worked well.

Here's more details:
iOS: 10.3.2
Device: iPhone 6
NRF iOS SDK version: 3.2.0

DFU Service Initiator parameters:
force_dfu = false
packetReceiptNotificationParameter = 12
enableUnsafeExperimentalButtonlessServiceInSecureDfu = true (didn't work with 'false' as well)

NRF LOG:
RF DFU Log (2017-09-11 14:17:26 +0000) : Connecting to Dfu52E5...
NRF DFU Log (2017-09-11 14:17:26 +0000) : centralManager.connect(peripheral, options: nil)
NRF DFU Log (2017-09-11 14:17:26 +0000) : [Callback] Central Manager did connect peripheral
NRF DFU Log (2017-09-11 14:17:26 +0000) : Connected to Dfu52E5
NRF DFU Log (2017-09-11 14:17:26 +0000) : Discovering services...
NRF DFU Log (2017-09-11 14:17:26 +0000) : peripheral.discoverServices(nil)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Services discovered
NRF DFU Log (2017-09-11 14:17:27 +0000) : Starting Legacy DFU...
NRF DFU Log (2017-09-11 14:17:27 +0000) : Connected to Dfu52E5
NRF DFU Log (2017-09-11 14:17:27 +0000) : Services discovered
NRF DFU Log (2017-09-11 14:17:27 +0000) : Legacy DFU Service found
NRF DFU Log (2017-09-11 14:17:27 +0000) : Discovering characteristics in DFU Service...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
NRF DFU Log (2017-09-11 14:17:27 +0000) : DFU characteristics discovered
NRF DFU Log (2017-09-11 14:17:27 +0000) : Reading DFU Version number...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
NRF DFU Log (2017-09-11 14:17:27 +0000) : Version number read: 0.8
NRF DFU Log (2017-09-11 14:17:27 +0000) : Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:27 +0000) : DFU Control Point notifications enabled
NRF DFU Log (2017-09-11 14:17:27 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Writing image sizes (0b, 0b, 38252b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:27 +0000) : peripheral.writeValue(0x00000000000000006c950000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
NRF DFU Log (2017-09-11 14:17:27 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:27 +0000) : Start DFU (Op Code = 1, Upload Mode = 4) request sent
NRF DFU Log (2017-09-11 14:17:29 +0000) : Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
NRF DFU Log (2017-09-11 14:17:29 +0000) : Response (Op Code = 1, Status = 1) received
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing Initialize DFU Parameters...
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:29 +0000) : peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:29 +0000) : peripheral.writeValue(0xffffffffffffffff0100feffc146, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
NRF DFU Log (2017-09-11 14:17:29 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:29 +0000) : peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:29 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
NRF DFU Log (2017-09-11 14:17:30 +0000) : Response (Op Code = 2, Status = 1) received
NRF DFU Log (2017-09-11 14:17:30 +0000) : Initialize DFU Parameters completed
NRF DFU Log (2017-09-11 14:17:30 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:30 +0000) : peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:30 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
NRF DFU Log (2017-09-11 14:17:30 +0000) : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
NRF DFU Log (2017-09-11 14:17:30 +0000) : peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
NRF DFU Log (2017-09-11 14:17:30 +0000) : Data written to 00001531-1212-EFDE-1523-785FEABCD123
NRF DFU Log (2017-09-11 14:17:30 +0000) : Uploading firmware...
NRF DFU Log (2017-09-11 14:17:30 +0000) : Sending firmware to DFU Packet characteristic...
NRF DFU Log (2017-09-11 14:17:43 +0000) : [Callback] Central Manager did disconnect peripheral
NRF DFU Log (2017-09-11 14:17:43 +0000) : Disconnected by the remote device
NRF DFU ERROR DFUError : Device disconnected unexpectedly

mostafaberg, 2017-09-18:
interesting !, may i ask for your help with debugging ? can you check if the issue exists in version `3.1.0` or not ? this will help us pinpoint the issue

gkovelman, 2017-09-18:
I don't have the resources for this for the coming week

mostafaberg, 2017-09-18:
Ok let us know when you have done the tests, we cannot reproduce that here as it requires your exact setup.

Looveh, 2017-09-26:
I'm experiencing the same issue on iOS 11, lib v4.0.1 unfortunately

mostafaberg, 2017-09-26:
@Looveh Can you open up a new issue and put in all the logs? prefer to have issues handled on a case by case basis for better granularity 👍

gkovelman, 2017-10-03:
FYI, I tested it again with the version 4.0.2. same results.
I'll try next week with version 3.1.0.

mostafaberg, 2017-11-06:
@Looveh @gkovelman any updates on this ? how does things work with the latest release ?

Looveh, 2017-11-06:
@mostafaberg No issues on our side! Thanks for the great job!

mostafaberg, 2017-11-06:
awesome!, thanks for the confirmation :) (Post vacation cleanup so sorry for some delays here and there)

closed 2017-11-06 by mostafaberg