nordicsemi/IOS-DFU-Library #142: DFU Not Working with iOS 11

257kyle, 2017-09-21:
After upgrading to iOS 11, DFU no longer works. It worked fine on iOS 10. 

It returns "DFU Service Not Found". 

Running iOS 11 on iphone 6S
DFU Library Version 3.2.1, already tried 3.1.0, but same issue
packetReceiptNotificationParameter = 12
ForceDFU = false

* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x0000000100b88954 LinkaIOS`DFUManagerViewController.dfuError(error=deviceNotSupported, message="DFU Service not found", self=0x00000001268b4c00) at DFUManagerViewController.swift:781
    frame #1: 0x0000000100b88aec LinkaIOS`@objc DFUManagerViewController.dfuError(_:didOccurWithMessage:) at DFUManagerViewController.swift:0
    frame #2: 0x0000000102728bcc iOSDFULibrary`closure #1 in BaseDFUExecutor.peripheralDidDisconnect(self=0x00000001027b83e8) at DFUExecutor.swift:83
    frame #3: 0x000000010272883c iOSDFULibrary`thunk for @callee_owned () -> () at DFUExecutor.swift:0
    frame #4: 0x000000010349d49c libdispatch.dylib`_dispatch_call_block_and_release + 24
    frame #5: 0x000000010349d45c libdispatch.dylib`_dispatch_client_callout + 16
    frame #6: 0x00000001034a2050 libdispatch.dylib`_dispatch_main_queue_callback_4CF + 1192
    frame #7: 0x0000000183d5ff20 CoreFoundation`__CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
    frame #8: 0x0000000183d5dafc CoreFoundation`__CFRunLoopRun + 2012
    frame #9: 0x0000000183c7e2d8 CoreFoundation`CFRunLoopRunSpecific + 436
    frame #10: 0x0000000185b0ff84 GraphicsServices`GSEventRunModal + 100
    frame #11: 0x000000018d22a880 UIKit`UIApplicationMain + 208
    frame #12: 0x0000000100a0f48c myApp`main at AppDelegate.swift:21
    frame #13: 0x00000001837a256c libdyld.dylib`start + 4

mostafaberg, 2017-09-21:
This seems like a simple Caching issue on iOS, I don't think it's related to iOS 11, toggle bluetooth on/off a couple of times until it works, if that's not the case then please verify that your peripheral actually has a DFU service being broadcasted.

257kyle, 2017-09-21:
Hey thanks for the response! I've tested it on a couple phones. One is an iphone 6+ and one is an iphone 6s. We had them both working with iOS 10 and then after updating to iOS 11 dfu fails. 

One thing we noticed is that with iOS 11, the bluetooth cache is cleared only if you turn on and off bluetooth in the settings app (rather than just by swiping up from the bottom). That helped us get further, and now DFU is starting, but moments after it still fails. This seems to be a different issue than the original one. 

The message is no longer "DFU Service Not Found" but now "Operation failed". 

Here is the new stack trace: 
* thread #1, queue = ‘com.apple.main-thread’, stop reason = breakpoint 1.1
  * frame #0: 0x0000000100b2493c myApp`DFUManagerViewController.showDetailError(string=“Operation failed”, self=0x0000000104057800) at DFUManagerViewController.swift:499
    frame #1: 0x0000000100b28a5c myApp`DFUManagerViewController.dfuError(error=remoteLegacyDFUOperationFailed, message=“Operation failed”, self=0x0000000104057800) at DFUManagerViewController.swift:782
    frame #2: 0x0000000100b28aec myApp`@objc DFUManagerViewController.dfuError(_:didOccurWithMessage:) at DFUManagerViewController.swift:0
    frame #3: 0x00000001026c8bcc iOSDFULibrary`closure #1 in BaseDFUExecutor.peripheralDidDisconnect(self=0x0000000102758ce0) at DFUExecutor.swift:83
    frame #4: 0x00000001026c883c iOSDFULibrary`thunk for @callee_owned () -> () at DFUExecutor.swift:0
    frame #5: 0x000000010343d49c libdispatch.dylib`_dispatch_call_block_and_release + 24
    frame #6: 0x000000010343d45c libdispatch.dylib`_dispatch_client_callout + 16
    frame #7: 0x0000000103442050 libdispatch.dylib`_dispatch_main_queue_callback_4CF + 1192
    frame #8: 0x0000000183d5ff20 CoreFoundation`__CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
    frame #9: 0x0000000183d5dafc CoreFoundation`__CFRunLoopRun + 2012
    frame #10: 0x0000000183c7e2d8 CoreFoundation`CFRunLoopRunSpecific + 436
    frame #11: 0x0000000185b0ff84 GraphicsServices`GSEventRunModal + 100
    frame #12: 0x000000018d22a880 UIKit`UIApplicationMain + 208
    frame #13: 0x00000001009af48c myApp`main at AppDelegate.swift:21
    frame #14: 0x00000001837a256c libdyld.dylib`start + 4

Looveh, 2017-09-22:
I'm facing the same issue

mostafaberg, 2017-09-22:
@Looveh can you open up a new issue ?

mostafaberg, 2017-09-22:
@257kyle are you still experiencing issues ? if so please open up a new ticket as your main issue opened here is resolved :)

closed 2017-09-22 by mostafaberg