nordicsemi/IOS-DFU-Library #143: Crashes during DFU on iOS 11

Looveh, 2017-09-22:
I had a project where DFU worked flawlessly on iOS 10.3.2 with lib version 3.2.1, but since upgrading to iOS 11 I get the following error:

```
Thread 6: EXC_BAD_ACCESS (code=1, address=0xe0d0c0b0a0908)
```

```
Thread 6 Queue : com.mycom.MyProd (serial)
#0	0x000e0d0c0b0a0908 in 0xe0d0c0b0a0908 ()
#1	0x00000001055bb904 in specialized static Zip.unzipFile(_:destination:overwrite:password:progress:fileOutputHandler:)
#2	0x00000001055b3f3c in static Zip.unzipFile(_:destination:overwrite:password:progress:fileOutputHandler:) ()
#3	0x000000010550299c in specialized static ZipArchive.unzip(_:)
#4	0x00000001054e45fc in static ZipArchive.unzip(_:) [inlined] ()
#5	0x00000001054e45f0 in specialized DFUStreamZip.init(urlToZipFile:type:)
#6	0x00000001054d1e80 in DFUStreamZip.init(urlToZipFile:type:) [inlined] ()
#7	0x00000001054d1e6c in DFUStreamZip.__allocating_init(urlToZipFile:type:) [inlined] ()
#8	0x00000001054d1e6c in DFUFirmware.init(urlToZipFile:type:)
#9	0x00000001054d1b54 in DFUFirmware.init(urlToZipFile:) [inlined] 
...
´´´ 

It seems to have something to do with the unzipping of the firmware file but I'm not sure what might cause the issue.

Looveh, 2017-09-22:
I tried running an older binary that was built with a Base SDK prior to 11.0 and that still worked on an iOS 11 device, so it likely doesn't work due to the upgrading to Base SDK 11.0.

mostafaberg, 2017-09-22:
you are absolutely right, there has been an update (yesterday?) for the Zip library to support iOS 11 and we did not get time to include that yet, but if you want to do that manually you may fork the repo and do a pod update to `Zip 1.0` if it is an emergency, if not we will push out an update soon for the DFU Library that has full iOS 11 support. until then i'll leave this issue open and keep you updated once it's out

damacster, 2017-09-22:
Thanks @mostafaberg. There are a couple of Swift warnings generated when working with `iOSDFULibrary` in Xcode 9.  Looking forward to the update.

257kyle, 2017-09-25:
Same here. Looking forward to the iOS 11 update.

mostafaberg, 2017-09-25:
Working on it :), will be up and running very soon !

mostafaberg, 2017-09-25:
4.0.0 is out now, thanks for your patience :)

closed 2017-09-25 by mostafaberg

damacster, 2017-09-25:
Thanks @mostafaberg!

Looveh, 2017-09-26:
Awesome job @mostafaberg !

mostafaberg, 2017-09-26:
Thanks for the cheers guys ! 💯 , let's keep up the good work 😄