nordicsemi/IOS-DFU-Library #145: iOS 11 Disconnects During DFU

257kyle, 2017-09-25:
Looks like DFU still fails with message "Operation failed". It worked on iOS 10, but with iOS 11 it disconnects immediately upon starting DFU. 

I tried updating to the latest commit (version 4.0.0) but the issue still remains. Here's the call stack: 

> * thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
>   * frame #0: 0x00000001008849ac myApp`DFUManagerViewController.showDetailError(string="Operation failed", self=0x000000014282f000) at DFUManagerViewController.swift:500
>     frame #1: 0x00000001008889bc myApp`DFUManagerViewController.dfuError(error=remoteLegacyDFUOperationFailed, message="Operation failed", self=0x000000014282f000) at DFUManagerViewController.swift:782
>     frame #2: 0x0000000100888a4c myApp`@objc DFUManagerViewController.dfuError(_:didOccurWithMessage:) at DFUManagerViewController.swift:0
>     frame #3: 0x000000010244cb8c iOSDFULibrary`closure #1 in BaseDFUExecutor.peripheralDidDisconnect(self=0x00000001024dcce0) at DFUExecutor.swift:83
>     frame #4: 0x000000010244c7fc iOSDFULibrary`thunk for @callee_owned () -> () at DFUExecutor.swift:0
>     frame #5: 0x00000001031c149c libdispatch.dylib`_dispatch_call_block_and_release + 24
>     frame #6: 0x00000001031c145c libdispatch.dylib`_dispatch_client_callout + 16
>     frame #7: 0x00000001031c6050 libdispatch.dylib`_dispatch_main_queue_callback_4CF + 1192
>     frame #8: 0x0000000181f73f20 CoreFoundation`__CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
>     frame #9: 0x0000000181f71afc CoreFoundation`__CFRunLoopRun + 2012
>     frame #10: 0x0000000181e922d8 CoreFoundation`CFRunLoopRunSpecific + 436
>     frame #11: 0x0000000183d23f84 GraphicsServices`GSEventRunModal + 100
>     frame #12: 0x000000018b43e880 UIKit`UIApplicationMain + 208
>     frame #13: 0x000000010070f3f0 myApp`main at AppDelegate.swift:21
>     frame #14: 0x00000001819b656c libdyld.dylib`start + 4
>

mostafaberg, 2017-09-25:
Check your caches again, this doesn't seem to be an issue with the actual library, are you testing on two different devices one with iOS 10 and one with iOS 11 ?

257kyle, 2017-09-25:
Wow, quick reply! 
Bluetooth cache isn't the issue (I'm toggling it in Settings). I've confirmed that it's an iOS 11 thing. I'm testing on two phones (one iOS 10.3.3 and the other iOS 11, both iphone 6+) with the same build, and it only works on one.

mostafaberg, 2017-09-25:
Yes but you need to toggle a few times, sometimes it won't work on the first toggle, also if you had any previous pairing information on the phone it needs to be removed from the iPhone's settings. if it works on one and not the other I can guarantee it's simply a cache issue, @philips77 , any tips ?

257kyle, 2017-09-25:
Yeah, we don't use pairing anymore, and there are zero paired bluetooth devices. 

Anyone else having iOS 11 issues? I assume you aren't able to repro it?

mostafaberg, 2017-09-25:
I'm not unfortunately, do you have a firmware that's compatible with one of our DKs that we can try ?

Looveh, 2017-09-26:
I'm experiencing the same issue on iOS 11, lib v4.0.1, my logs are
```
2017-09-26 11:13:41.321000+0200 NordicDFUExample[1606:2106258] Starting DFU
2017-09-26 11:13:41.359795+0200 NordicDFUExample[1606:2106245] logWith: 1 message: 'Connecting to PBX-65014E2024A9B36D...'
2017-09-26 11:13:41.359939+0200 NordicDFUExample[1606:2106245] logWith: 0 message: 'centralManager.connect(peripheral, options: nil)'
2017-09-26 11:13:41.613922+0200 NordicDFUExample[1606:2106209] logWith: 0 message: '[Callback] Central Manager did connect peripheral'
2017-09-26 11:13:41.614246+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Connected to PBX-65014E2024A9B36D'
2017-09-26 11:13:41.614460+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Discovering services...'
2017-09-26 11:13:41.614688+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.discoverServices(nil)'
2017-09-26 11:13:42.100914+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Services discovered'
2017-09-26 11:13:42.101754+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Starting Secure DFU...'
2017-09-26 11:13:42.102686+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Connected to PBX-65014E2024A9B36D'
2017-09-26 11:13:42.102942+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Services discovered'
2017-09-26 11:13:42.103157+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Secure DFU Service found'
2017-09-26 11:13:42.104490+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Discovering characteristics in DFU Service...'
2017-09-26 11:13:42.104778+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.discoverCharacteristics(nil, for: FE59)'
2017-09-26 11:13:42.154656+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'DFU characteristics discovered'
2017-09-26 11:13:42.155670+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2017-09-26 11:13:42.156012+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)'
2017-09-26 11:13:42.273218+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50'
2017-09-26 11:13:42.273523+0200 NordicDFUExample[1606:2106209] logWith: 10 message: 'Buttonless DFU indications enabled'
2017-09-26 11:13:42.273787+0200 NordicDFUExample[1606:2106209] logWith: 15 message: 'Application with buttonless update found'
2017-09-26 11:13:42.274614+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Trying setting bootloader name to Dfu57912'
2017-09-26 11:13:42.275218+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2017-09-26 11:13:42.276352+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.writeValue(0x02084466753537393132, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)'
2017-09-26 11:13:42.332051+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50'
2017-09-26 11:13:42.332897+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200202'
2017-09-26 11:13:42.333031+0200 NordicDFUExample[1606:2106209] logWith: 20 message: 'Error 2: Operation not supported'
2017-09-26 11:13:42.333139+0200 NordicDFUExample[1606:2106209] logWith: 15 message: 'Setting bootloader name not supported'
2017-09-26 11:13:42.333262+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2017-09-26 11:13:42.333375+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)'
2017-09-26 11:13:42.393000+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50'
2017-09-26 11:13:42.394058+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101'
2017-09-26 11:13:42.394522+0200 NordicDFUExample[1606:2106209] logWith: 10 message: 'Response (Op Code = 1, Status = 1) received'
2017-09-26 11:13:43.156443+0200 NordicDFUExample[1606:2106209] logWith: 0 message: '[Callback] Central Manager did disconnect peripheral'
2017-09-26 11:13:43.156733+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Disconnected by the remote device'
2017-09-26 11:13:43.157070+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Scanning for the DFU Bootloader...'
```

mostafaberg, 2017-09-26:
@Looveh Your issue from the logs seems different than this one you bumped, can you open a separate issue please ?

Looveh, 2017-09-26:
@mostafaberg Sure, sorry for the inconvenience, opened new issue here #149

257kyle, 2017-09-27:
Hey @mostafaberg we were able to fix it by setting packetReceiptNotificationParameter to 1. (We had it at 12). Thanks for the help!

closed 2017-09-27 by 257kyle

mostafaberg, 2017-09-27:
Perfect, I would say you can go a bit higher than 1 since this will make things a bit slow, you can try to set it to 6, 8, etc.. until you see the error reproduce, then lower it by 2 and this can be your new safe value. the minimum guaranteed value to work is 2, some peripherals can handle PRN higher than 12, but seems like yours will do well in the 6-8 range