nordicsemi/IOS-DFU-Library #149: v4.0.1 Disconnects during DFU

Looveh, 2017-09-26:
I'm experiencing issues where the DFU process is halted right after being initiated.

OS: 11
Library version: 4.0.1
Language: Objective C

My setup was confirmed to work with iOS 10.3.2 and lib v3.2.1.

Logs: 

```
2017-09-26 11:13:41.321000+0200 NordicDFUExample[1606:2106258] Starting DFU
2017-09-26 11:13:41.359795+0200 NordicDFUExample[1606:2106245] logWith: 1 message: 'Connecting to PBX-65014E2024A9B36D...'
2017-09-26 11:13:41.359939+0200 NordicDFUExample[1606:2106245] logWith: 0 message: 'centralManager.connect(peripheral, options: nil)'
2017-09-26 11:13:41.613922+0200 NordicDFUExample[1606:2106209] logWith: 0 message: '[Callback] Central Manager did connect peripheral'
2017-09-26 11:13:41.614246+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Connected to PBX-65014E2024A9B36D'
2017-09-26 11:13:41.614460+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Discovering services...'
2017-09-26 11:13:41.614688+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.discoverServices(nil)'
2017-09-26 11:13:42.100914+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Services discovered'
2017-09-26 11:13:42.101754+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Starting Secure DFU...'
2017-09-26 11:13:42.102686+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Connected to PBX-65014E2024A9B36D'
2017-09-26 11:13:42.102942+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Services discovered'
2017-09-26 11:13:42.103157+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Secure DFU Service found'
2017-09-26 11:13:42.104490+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Discovering characteristics in DFU Service...'
2017-09-26 11:13:42.104778+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.discoverCharacteristics(nil, for: FE59)'
2017-09-26 11:13:42.154656+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'DFU characteristics discovered'
2017-09-26 11:13:42.155670+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2017-09-26 11:13:42.156012+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)'
2017-09-26 11:13:42.273218+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50'
2017-09-26 11:13:42.273523+0200 NordicDFUExample[1606:2106209] logWith: 10 message: 'Buttonless DFU indications enabled'
2017-09-26 11:13:42.273787+0200 NordicDFUExample[1606:2106209] logWith: 15 message: 'Application with buttonless update found'
2017-09-26 11:13:42.274614+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Trying setting bootloader name to Dfu57912'
2017-09-26 11:13:42.275218+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2017-09-26 11:13:42.276352+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.writeValue(0x02084466753537393132, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)'
2017-09-26 11:13:42.332051+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50'
2017-09-26 11:13:42.332897+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200202'
2017-09-26 11:13:42.333031+0200 NordicDFUExample[1606:2106209] logWith: 20 message: 'Error 2: Operation not supported'
2017-09-26 11:13:42.333139+0200 NordicDFUExample[1606:2106209] logWith: 15 message: 'Setting bootloader name not supported'
2017-09-26 11:13:42.333262+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2017-09-26 11:13:42.333375+0200 NordicDFUExample[1606:2106209] logWith: 0 message: 'peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)'
2017-09-26 11:13:42.393000+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50'
2017-09-26 11:13:42.394058+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101'
2017-09-26 11:13:42.394522+0200 NordicDFUExample[1606:2106209] logWith: 10 message: 'Response (Op Code = 1, Status = 1) received'
2017-09-26 11:13:43.156443+0200 NordicDFUExample[1606:2106209] logWith: 0 message: '[Callback] Central Manager did disconnect peripheral'
2017-09-26 11:13:43.156733+0200 NordicDFUExample[1606:2106209] logWith: 5 message: 'Disconnected by the remote device'
2017-09-26 11:13:43.157070+0200 NordicDFUExample[1606:2106209] logWith: 1 message: 'Scanning for the DFU Bootloader...'
```

mostafaberg, 2017-09-26:
Does this update only flash the application firmware ? or is it also updated the softdevice ?

Looveh, 2017-09-26:
@mostafaberg Only application, not SD

piellarda, 2017-09-26:
I'm having a similar issue but I don't have as much log.
The process get stuck after the disconnection, when the device has jumped to bootloader.
I can make the update if I retry the process from bootloader but the reconnection fails again when the device has jump back to our firmware.
It's working fine if I downgrade the SDK to 3.2.1.
I tried on iPhone 6s (iOS 10.3.3) and iPhone 7 (11.0).

philips77, 2017-09-26:
Are you sure your bootloader advertises with dfu service uuid after switching from application? There are 3 options:
Your bootloader may advertise with some other data, it may failed to start, the lib may have error flag set after failing setting name (SDK 14 feature).

Looveh, 2017-09-27:
@philips77 Yes, when the DFU process is halted the device is still in DFU mode and I'm able to find it in the Nordic nRF Connect app where it's advertising on the DFU service. 

If it's any help, we're using nRF SDK v13 on the device.

Looveh, 2017-09-28:
Tried setting `packetReceiptNotificationParameter` to 1 and 2 (used to be default 12) without any improvement.

Looveh, 2017-09-28:
Our firmware uses nRF SDK 13

piellarda, 2017-09-28:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/3b2c99ec335fcc3c9e9680eff6f857bcfacbd80a/iOSDFULibrary/Classes/Implementation/DFUPeripheralSelector.swift#L33
Here's the problem, this used to return true before 4.0.0 and since the condition above always fails for us, the device is never found.

philips77, 2017-09-28:
Of course is should return true, like the old method did before. We will fix it right now. Sorry!

mostafaberg, 2017-09-28:
@piellarda  @Looveh Thansk for the feedback, that would be my mistake, will fix that shortly ! thanks for the help !

mostafaberg, 2017-09-28:
Issue now resolved !, please try 4.0.2 and confirm :)

piellarda, 2017-09-28:
All good for me, thanks.

mostafaberg, 2017-09-28:
Perfect!, what abou t @Looveh ? does 4.0.2 resolve your issue ?

Looveh, 2017-09-28:
4.0.2 works for me too. Awesome job @mostafaberg, @philips77! Don't think I've ever come across a project with such high quality and fast feedback, thanks as always!

closed 2017-11-24 by philips77